f4exb/sdrangel #877: Possible memory leak in Satellite Tracker

f4exb, 2021-04-28:
There is a possible memory leak in the Satellite Tracker feature plugin. Here is the memory usage graph before and after restarting SDRangel. Before with the plugin and after without:
[image]
This is the instance of SDRangel that I routinely use and let run to monitor the local UHF and VHF repeaters traffic. It has the same setup but before restart I had added the Satellite Tracker to occasionally keep an eye on what's going on there. This seems to designate a possible memory leak issue in the Satellite Tracker.

srcejon, 2021-04-28:
I'll take a look - did you have the map open?

srcejon, 2021-04-28:
I can see it rising without the map. Perhaps this was the problem when I thought there might be a leak in the APT demod.

srcejon, 2021-04-28:
I see one leak. Easy enough to fix.

f4exb, 2021-04-28:
OK good! And I confirm that no map was open.

f4exb, 2021-04-28:
looks like a fix. I had a doubt but this was due to Zoom and Firefox. I'll keep watching a little bit.

f4exb, 2021-04-29:
I think it is OK now:
[image]

f4exb, 2021-05-01:
Deployed in v6.10.3

closed 2021-05-01 by f4exb