f4exb/sdrangel #878: SDRangel Not REsponding MacOS

benb0jangles, 2021-04-28:


benb0jangles, 2021-04-28:
Ok, so my CubicSDR installation also hangs with rtlsdr, so I think it is something wrong with:
`2021-04-28 14:56:43.476 (I) PluginManager::loadPluginsDir: loaded plugin libinputrtlsdr.dylib`

Does anybody know how to remove rtlsdr from Soapysdr? Or do I have to remove it all and redo?

benb0jangles, 2021-04-29:
Solved:

First install splash screen freeze:
SDRangel 6.10.1

Second install same result:
SDRangel 6.10.2

Working with:
SDRangel 6.6.0

Note:
Using my standard RTL-SDR device works, but Airspy Mini - it show in sdrangel in device list but play button doesn't start waterfall.

iMac 10.12.6 Sierra
SoapySDR, version 0.8.0
installed SDRangel with:

```
git clone --single-branch https://github.com/macports/macports-ports.git
cd macports-ports
git checkout (add commit number here)
sudo port install
```

ra1nb0w, 2021-05-18:
This is very strange. SoapyRTLSDR was not updated recently (last 2020-02-29) like rtl-sdr (last 2021-01-08).
@benb0jangles can you try to install a version before rtl-sdr @20201031-0847e93e? thank you

To remove rtlsdr from soapy you have the following solution:
```
sudo port uninstall SoapyRTLSDR
sudo port install sdrangel -rtlsdr
```

cfc62, 2021-06-11:
I am running into a very similar if not the same issue.

Install via MacPorts - no errors or messages of note

Run - also no errors or messages of note, including console output

However, while the SDR device is seen, and can be selected (Airspy in this case), nothing happens and no waterfall starts.

This applies to both 16.12.1 and 6.13.0.

ra1nb0w, 2021-06-12:
Are you using the native airspy interface or via soapysdr?

cfc62, 2021-06-12:
I don't see the soapysdr interface shown in the drop-down source selection list; I do see:

Airspy[0] <serial number>

Where <serial number> is the actual s/n of the connected Airspy device.

ra1nb0w, 2021-06-12:
Ok. this seems a different kind of problem. Can you try to use gqrx?

cfc62, 2021-06-13:
gqrx works great, no problems at all.

cfc62, 2021-07-25:
Must have been an OS-related bug, as updating to 11.5 and making no other changes whatsoever fixes the issue.

ra1nb0w, 2021-07-25:
airspy doesn't work on gqrx or cubicsdr?

closed 2021-12-19 by github-actions[bot]