f4exb/sdrangel #881: DAB/DAB + demodulator v6.10.3

choupette-change, 2021-05-02:
version Windows ok
version .deb Ubuntu 20 10 : blocking of control buttons and/or immediate crash ?????

2021-05-02 13:47:41.880 (D) DABDemodSink::applyChannelSettings:  channelSampleRate:  2048000  channelFrequencyOffset:  0
2021-05-02 13:47:41.922 (W) QOpenGLTexture::setData() tried to set a null image
--Type <RET> for more, q to quit, c to continue without paging--

Thread 48 "sdrangel" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff91380640 (LWP 4628)]
__memmove_avx_unaligned_erms () at ../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S:248
248	../sysdeps/x86_64/multiarch/memmove-vec-unaligned-erms.S: Aucun fichier ou dossier de ce type.
(gdb)

srcejon, 2021-05-03:
Can you try typing 'info stack' at the gdb prompt, please?

Does it always crash? Or just under a specific set of conditions?

choupette-change, 2021-05-03:
2021-05-03 18:57:02.690 (D) DABDemodSink::applyChannelSettings:  channelSampleRate:  2048000  channelFrequencyOffset:  0
2021-05-03 18:57:02.731 (W) QOpenGLTexture::setData() tried to set a null image
--Type <RET> for more, q to quit, c to continue without paging--

Thread 46 "sdrangel" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fff91380640 (LWP 6148)]
0x00007ffff7fd988b in ?? () from /lib64/ld-linux-x86-64.so.2
(gdb) info stack
#0  0x00007ffff7fd988b in  () at /lib64/ld-linux-x86-64.so.2
#1  0x00007ffff7fda479 in  () at /lib64/ld-linux-x86-64.so.2
#2  0x00007ffff7fdf3a7 in  () at /lib64/ld-linux-x86-64.so.2
#3  0x00007ffff7fe6eae in  () at /lib64/ld-linux-x86-64.so.2
#4  0x00007fffb4631aa6 in sampleReader::getSample(int) ()
    at /usr/local/lib/libdab_lib.so
#5  0x00007fffb46320b2 in timeSyncer::sync(int, int) ()
    at /usr/local/lib/libdab_lib.so
#6  0x00007fffb462f310 in dabProcessor::run() ()
    at /usr/local/lib/libdab_lib.so
#7  0x00007ffff5fe7d84 in  () at /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff5390590 in start_thread (arg=0x7fff91380640)
    at pthread_create.c:463
#9  0x00007ffff7ed2223 in clone ()
    at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) 

----> Does it always crash? Or just under a specific set of conditions?
when launching the demodulator (blue button) each time. Ok with the other demodulator ??

closed 2021-11-25 by github-actions[bot]