f4exb/sdrangel #882: DSD demodulator

gaspar1987, 2021-05-03:
Bonjour au créateur sdrangel Accro de DSD et je voudrais qu'il soit amélioré dans la version 1 9 3 avec de la crypto audio pour mieux pour que ça marche avec le talkie-walkie avec  ceci et mieux pour la création du audio j'aime bien ce plugin DMR  c'est pour ça que je voudrais que vous l'améliorer pour que je puisse améliorer la réception des trames parce que c'est un peu écouter même avec de la crypto mais en automatique ça serait mieux pour il se branche sur la fréquence parce que avec le mode manuel ça ne fonctionne pas j'ai testé plusieurs fois avec les anciens dérange elle de la version ancienne .
        Bonne journée

gaspar1987, 2021-05-03:
[image]

f4exb, 2021-05-03:
Sorry Basic Privacy will never be supported. Period.

closed 2021-05-03 by f4exb