f4exb/sdrangel #883: Lora ChirpChat Demodulator returns uncorrect data.

zskdan, 2021-05-03:
Hello,

I'm using the latest available version of sdrangel (6.10.3). But I cannot  make the ChirpChat Demodulator with plutoSDR to demod some Lora packet that I sent using an RFM95 module from an arduino.

I've set correctly Lora parameters to match transmitter configuration (SF=7, Pre=8, DE=2 (not sure) and FEC=1, BW=125khz, CRC=OFF and HDR=OFF). But no luck :(

My Transmitter send a payload containing this String: `HeLoRa World!`

`
 48 65 4c 6f 52 61 20 57  6f 72 6c 64 21 0a        |HeLoRa World!.
`

But I got the following from ChirpChat: 
`>17:36:36 55 S:-30.0 SN:-6.3 HF:err HC:err FEC:fix CRC:err
000|fe ec 9a 71|0b cd ea bb|d5 21 9a 5b|29 f4 94 6c
016|5a 33 1d b4|c5 c7 6a 21|e3 3a 5c 03|08 
TXT|���q����!�[)��lZ3���j!�:
`
Using a second module (based on RN2483 module) I got it successfully.

Am I doing something wrong?
Thanks for your support!

f4exb, 2021-05-03:
DE or Distance Enhancement (not very descriptive) combines adjacent FFTs to make one symbol so it is a very important parameter. SF=7 and DE=2 makes only 2⁵ = 32 possible symbols and not 2⁷ = 128 if SF was 0.

If Spreading Factor (SF) was 12 then DE=2 is mandatory but for SF=7 it is not so sure. Can you try with DE=0 ? In LoRa the only possible values for DE is either 0 or 2.

f4exb, 2021-05-04:
These many errors are also an indication of a mismatch between Tx and Rx;
`17:36:36 55 S:-30.0 SN:-6.3 HF:err HC:err FEC:fix CRC:err`

Also a S/N ratio of -6.3 is not good considering this is the S/N of the de-chirped signal. In these conditions it is no wonder that you decode garbage.

zskdan, 2021-05-05:
Since I can change settings on transmitter. I've changed the SF to 12 and DE to 2. And in-fact I got almost 90% of my Lora packets demod-ed in ChirpChat.

So maybe I've missed something on the SF==7, I'll doublecheck in the next few days.

Thanks for your support!

zskdan, 2021-05-07:
I Just retest with SF==7, 8, 9 and unfortunately I keep having garbage whatever DE value is 0 or 2.

On the other hand, SF 11 and 12 works great. I demod-ed almost all my packets. 

I can stick on SF12, so feel free to close this issue, otherwise If you consider that this should be clarified more further, I'll be more than happy to provide more investigation/details.

f4exb, 2021-05-07:
In fact SF=11 or 12 with DE=2 are the most sensitive modes so maybe thre is an issue with decoding faster but more fragile modes. Can you please tell me what are the other fundamental parameters: bandwidth and number of preamble chirps. These parameters match the (3) and (10) controls here: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/demodchirpchat#readme

zskdan, 2021-05-08:
I'm using Preamble at 8 and the bandwidth at 125khz.

f4exb, 2021-06-12:
I think this is a limitation that would be very difficult to overcome so I am not going to do it. One does have to make FFT collation (i.e. DE not zero) so for LoRa it means SF=11 or 12. This will be reflected in the readme.

f4exb, 2021-06-12:
Readme updated

closed 2021-06-12 by f4exb