f4exb/sdrangel #884: Spurs on transmit when using interpolation of 64

sh123, 2021-05-04:
When trying to transmit tone and using interpolation of 64 output spectra has lots of spurs, when using interpolation of 32 and less spurs do not exist.

[image]

f4exb, 2021-05-04:
Any idea of the level of these spurs? It is normal to have spurs since in digital technology you cannot perform calculations with infinite precision.

sh123, 2021-05-04:
@f4exb , some people complained about it when tried to transmit over QO-100, so yes, they are noticeable, but looks like they are visible when using interpolation of 64.

f4exb, 2021-05-04:
"visible", "noticeable"... is not a measurement.

sh123, 2021-05-04:
[image]

f4exb, 2021-05-04:
OK it seems consistent with what is observed using the TestSink plugin (thus just pure "computation"). Modulating signal is an unmodulated AM CW signal (-6dB) it has no artifacts even down to -100 dBc:
[image]

Output at interpolation of 32 (and down to 32 it is about the same). Spurs are about -80 dBc thus perfectly acceptable
[image]

Now at interpolation of 64 spurs are up to about -20 dBc close to the carrier and significantly widespread all over the band. Indeed this is not normal.
[image]

sh123, 2021-05-04:
Yes, also did some better measurement

32 seems to be OK

[image]

64 has some spurs

[image]

f4exb, 2021-05-04:
The waveform is indeed consistent with what is sent to the device. There is some filtering in hardware that reshapes it a bit but the many spurs in what looks like a sinc function are clearly visible. I guess there is a bug in the calculation that does some wrong sample swapping somewhere. I'll have a closer look.

f4exb, 2021-05-04:
It is obviously missing the offsets from +110 to +127 at the end of the iteration here: https://github.com/f4exb/sdrangel/blob/master/sdrbase/dsp/interpolators.h#L1238

closed 2021-05-04 by f4exb

sh123, 2021-05-05:
@f4exb , yes, works fine now, clean signal on qo-100, thank you.

commit b04cc965 (2021-05-04): Fixed center interpolator by 64 missing some code. Fixes #884