f4exb/sdrangel #887: Tag / Bookmark frequency

luxwarp, 2021-05-05:
Hey! The feature of saving presets works as some kind of bookmark feature and thats great, but i would really like a small feature to be able to tag a frequency and then a label should be shown in the spectrum. Just to be able to label let's say 118.600 Mhz - Landvetter Twr.

Just to be able to label a frequency, no need to save demodulator settings etc like saving presets.

f4exb, 2021-05-05:
This was already requested in a long list of features here #768 Many of them were implemented but not this one. I was a bit reluctant because it may require screen real estate re-organziation but may be not necessary depending on how this is implemented. This is the quote of the original request by `srcejon`:

> With regards to labels, it's not a case of implementing them because other software has them - but I think more that other s/w has them because they are useful. A couple of use cases: it's useful to easily see what modes should be used within subbands of the HAM bands and where the band limits are. Also, it's useful for labelling things like beacons, so you easily know what you are looking at. Some may be able to remember the frequencies in their head, but I have to continually look them up, so it's nice just to have them on screen

So due to popular demand I think we will have to implement this.

f4exb, 2021-05-05:
Thinking of something along these lines not with a "channel" of course but with the same sort of idea:
[image]

I think one more line will have to be inserted below the present channel display which will reduce the spectrum useful display but this is the price to pay.

luxwarp, 2021-05-05:
Ah that's exactly what I was thinking about ! Maybe you could implement a really basic function of this to get it started and then we could build it further with more options and settings, like import, export, groups/categories, connections to presets, colors etc .

Yes of course you would loose some viewable display but a simple on/off switch could be useful. 

And for the record, sdrangel is awesome and all your work is amazing. All features, demodulators, decoders, etc, all in one place , that is what will make sdrangel the go to SDR application in the future for many users !

f4exb, 2021-05-05:
"connection to presets" well not exactly... These markers have to be tied to a spectrum and there are many more spectrum displays than just the big baseband spectrum display in the middle. So I see it tied to the spectrum settings. However I think these spectrum settings are also part of the various presets including those for the channels. Because this is a spectrum thing the control of these markers has to be done from the spectrum GUI.

srcejon, 2021-05-05:
Looks good.

I would have thought they would be less useful in the other spectrum windows, so probably hidden by default (so there is no lost space)?

There's perhaps another meaning of 'connection to presets'. You could perhaps allow a preset to be associated with a marker - then double clicking the marker loads the preset (possibly with the frequency of the preset being overridden by the frequency of the marker).

f4exb, 2021-05-06:
Note that spectrum overlay is also possible like with the present spectrum markers:
[image]

f4exb, 2021-05-06:
There will be 3 types of marker overlays possible:
  - the channel markers: no change
  - the spectrum markers: the analyzer type of marker similar to what exists today but made a marker on its own right following the frequency and thus possibly move the power level display accordingly. This type of marker attached to a spectrum could possibly be saved.
  - the new "anotation" markers new to this ticket. Similarly to channel markers they also have a bandwidth (one could specify start and stop rather than center and bandwidth) and may be displayed as an overlaying bar at the bottom of the spectrum with the annotation text on one side. These markers could be piled up from largest in bandwidth to smallest. Thus the most specific are always displayed.

Of course one may not want to see all this all the time. While the channel markers can keep their functionality as it is today the spectrum and annotation markers could be optional and exclusive. Generally one would need one or the other to do either instrumentation or band scanning.

I do not really understand why a preset would be associated to a marker. Generally one would load a set of markers in a particular spectrum. The annotation markers could be saved in a .csv file (a la SdrConsole). It is up to the user to make sure the file exists else nothing is displayed without disruption of main functionality. The file path to the annotations description is then maintained in the spectrum settings of the preset.

This functionnality being attached to the spectrum component it is maintained wherever the spectrum component is used and whether this makes sense or not (main or channel displays). In some particular cases one may want to store annotations as calibration marks. Later it could be possible to store spectrum templates this way.

f4exb, 2022-01-23:
Implemented in v6.18.1

closed 2022-01-23 by f4exb

srcejon, 2022-01-24:
Lovely - thanks :)