f4exb/sdrangel #888: error at "make" upgrade Version: 6.10.2 to Version: 6.10.3

choupette-change, 2021-05-05:
tutu@tutu-Aspire-E5-573:/opt/build/sdrangel/build$ cmake -Wno-dev -DDEBUG_OUTPUT=ON -DRX_SAMPLE_24BIT=ON -DCMAKE_BUILD_TYPE=RelWithDebInfo -DMIRISDR_DIR=/opt/install/libmirisdr -DAIRSPY_DIR=/opt/install/libairspy -DAIRSPYHF_DIR=/opt/install/libairspyhf -DBLADERF_DIR=/opt/install/libbladeRF -DHACKRF_DIR=/opt/install/libhackrf -DRTLSDR_DIR=/opt/install/librtlsdr -DLIMESUITE_DIR=/opt/install/LimeSuite -DIIO_DIR=/opt/install/libiio -DPERSEUS_DIR=/opt/install/libperseus -DXTRX_DIR=/opt/install/xtrx-images -DSOAPYSDR_DIR=/opt/install/SoapySDR -DUHD_DIR=/opt/install/uhd -DAPT_DIR=/opt/install/aptdec -DCM256CC_DIR=/opt/install/cm256cc -DDSDCC_DIR=/opt/install/dsdcc -DSERIALDV_DIR=/opt/install/serialdv -DMBE_DIR=/opt/install/mbelib -DCODEC2_DIR=/opt/install/codec2 -DSGP4_DIR=/opt/install/sgp4 -DLIBSIGMF_DIR=/opt/install/libsigmf -DDAB_DIR=/opt/install/libdab -DCMAKE_INSTALL_PREFIX=/opt/install/sdrangel ..
-- SDRangel Version: 6.10.3
-- Build type set to: RelWithDebInfo
-- Compiling for 24 bit Rx DSP chain
-- Target architecture: x86_64-native
-- Looking for __AVX__ - forced found
-- Looking for __SSE4_2__ - forced found
-- Looking for __SSE4_1__ - forced found
-- Looking for __SSSE3__ - forced found
-- Looking for __SSE3__ - forced found
-- Looking for __SSE2__ - forced found
-- Interprocedural optimization enabled
-- Found FFTW3F: /usr/include, /usr/lib/x86_64-linux-gnu/libfftw3f.so, /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so
-- FAAD LIBRARY /usr/lib/x86_64-linux-gnu/libfaad.so
-- FAAD INCLUDE DIRS /usr/include
-- Checking for module 'sgp4' (already installed by source for 6.10.2 version??)
--   No package 'sgp4' found**
-- SGP4 LIBRARIES /opt/install/sgp4/lib/libsgp4s.so
-- SGP4 INCLUDE DIRS /opt/install/sgp4/include/SGP4
-- Checking for module 'apt'   (already installed by source for 6.10.2 version??)
--   No package 'apt' found
-- APT LIBRARIES /opt/install/aptdec/lib/libapt.so
-- APT INCLUDE DIRS /opt/install/aptdec/include/apt
-- Checking for module 'libairspyhf'
--   No package 'libairspyhf' found
-- libairspyhf not found.
-- Checking for module 'libbladeRF>=2.0'
--   No package 'libbladeRF' found
-- libbladerf not found.
-- Could not find Lime Suite
-- Checking for module 'libiio>=0.7'
--   No package 'libiio' found
-- libiio not found.
-- Checking for module 'libperseus'
--   No package 'libperseus' found
-- libperseus not found.
-- SDRPLAY LIBRARIES /usr/local/lib/libsdrplay_api.so
-- SDRPLAY INCLUDE DIRS /usr/local/include
-- Found SoapySDR: /opt/install/SoapySDR/include, /opt/install/SoapySDR/lib/libSoapySDR.so
-- Checking for module 'libxtrx'
--   No package 'libxtrx' found
-- libxtrx not found.
-- Checking for module 'uhd'
--   No package 'uhd' found
-- UHD LIBRARIES UHD_LIBRARIES-NOTFOUND
-- UHD INCLUDE DIRS UHD_INCLUDE_DIR-NOTFOUND
-- Could NOT find UHD (missing: UHD_LIBRARIES UHD_INCLUDE_DIR) 
-- Include demoddatv
-- added optional 'make dev_docs' for developer documentation generation
-- Check for case-sensitive file systems
-- File system is case-sensitive
-- Asking qmake for QT_PLUGINS_DIR and got /usr/lib/x86_64-linux-gnu/qt5/plugins
-- Asking qmake for QT_IMPORTS_DIR and got /usr/lib/x86_64-linux-gnu/qt5/imports
-- Asking qmake for QT_QML_DIR and got /usr/lib/x86_64-linux-gnu/qt5/qml
-- CPack generators: 
-- Configuring done
-- Generating done
-- Build files have been written to: /opt/build/sdrangel/build
tutu@tutu-Aspire-E5-573:/opt/build/sdrangel/build$ make -j4
make[1] : on entre dans le répertoire « /opt/build/sdrangel/build »
make[2] : on entre dans le répertoire « /opt/build/sdrangel/build »
make[2] : on entre dans le répertoire « /opt/build/sdrangel/build »
make[2] : on entre dans le répertoire « /opt/build/sdrangel/build »
make[2] : on entre dans le répertoire « /opt/build/sdrangel/build »
make[2] : on quitte le répertoire « /opt/build/sdrangel/build »
make[2] : on entre dans le répertoire « /opt/build/sdrangel/build »
make[2] : on quitte le répertoire « /opt/build/sdrangel/build »
make[2] : on entre dans le répertoire « /opt/build/sdrangel/build »
[  0%] Automatic MOC and UIC for target httpserver
.
.
.
[ 41%] Built target demodvorscsrv
[ 41%] Building CXX object pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/aptdemodbaseband.cpp.o
[ 41%] Building CXX object pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/aptdemodsink.cpp.o
[ 41%] Building CXX object pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/aptdemodimageworker.cpp.o
[ 41%] Building CXX object pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/aptdemodplugin.cpp.o
[ 41%] Building CXX object pluginssrv/channelrx/demodapt/CMakeFiles/demodaptsrv.dir/aptdemodwebapiadapter.cpp.o
[ 42%] Linking CXX shared library ../../../lib/pluginssrv/libdemodaptsrv.so
make[2] : on quitte le répertoire « /opt/build/sdrangel/build »
[ 42%] Built target demodaptsrv
make[1] : on quitte le répertoire « /opt/build/sdrangel/build »
make: *** [Makefile:152 : all] Erreur 2
tutu@tutu-Aspire-E5-573:/opt/build/sdrangel/build$

srcejon, 2021-05-05:
I think you've cut out the error message. Can you run make without -j

choupette-change, 2021-05-05:
without -j
[ 36%] Building CXX object plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/demoddab_autogen/mocs_compilation.cpp.o
cd /opt/build/sdrangel/build/plugins/channelrx/demoddab && /usr/bin/c++  -DAPPLICATION_NAME=\"SDRangel\" -DARCHITECTURE_x86_64 -DCOMPANY=\"f4exb\" -DCOPYRIGHT="\"Copyright (c) 2019 Edouard Griffiths. All rights reserved.\"" -DDABLIN_AAC_FAAD2 -DHAS_LIBSIGMF -DIDENTIFIER=\"com.f4exb.SDRangel\" -DLINUX -DQT_CORE_LIB -DQT_GUI_LIB -DQT_MULTIMEDIA_LIB -DQT_NETWORK_LIB -DQT_NO_DEBUG -DQT_OPENGL_LIB -DQT_WEBSOCKETS_LIB -DQT_WIDGETS_LIB -DSDRANGEL_LIB_VERSION=\"6.10.3\" -DSDRANGEL_VERSION=\"6.10.3\" -DSDR_RX_SAMPLE_24BIT -DUSE_AVX -DUSE_AVX2 -DUSE_SSE2 -DUSE_SSE3 -DUSE_SSE4_1 -DUSE_SSE4_2 -DUSE_SSSE3 -Ddemoddab_EXPORTS -I/opt/build/sdrangel/build/plugins/channelrx/demoddab -I/opt/build/sdrangel/plugins/channelrx/demoddab -I/opt/build/sdrangel/build/plugins/channelrx/demoddab/demoddab_autogen/include -I/opt/build/sdrangel/sdrbase -I/opt/build/sdrangel/exports -I/opt/build/sdrangel/sdrgui -I/opt/build/sdrangel/swagger/sdrangel/code/qt5/client -I/opt/install/libdab/include/dab_lib -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -isystem /usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtMultimedia -isystem /usr/include/x86_64-linux-gnu/qt5/QtNetwork -isystem /usr/include/x86_64-linux-gnu/qt5/QtWebSockets -isystem /usr/include/x86_64-linux-gnu/qt5/QtOpenGL  -O2 -g -DNDEBUG -flto -fno-fat-lto-objects -fPIC   -march=native -Wall -Wextra -Wvla -Woverloaded-virtual -ffast-math -ftree-vectorize -fPIC -std=c++14 -o CMakeFiles/demoddab.dir/demoddab_autogen/mocs_compilation.cpp.o -c /opt/build/sdrangel/build/plugins/channelrx/demoddab/demoddab_autogen/mocs_compilation.cpp
In file included from /opt/build/sdrangel/plugins/channelrx/demoddab/dabdemodbaseband.h:29,
                 from /opt/build/sdrangel/plugins/channelrx/demoddab/dabdemod.h:31,
                 from /opt/build/sdrangel/build/plugins/channelrx/demoddab/demoddab_autogen/EWIEGA46WW/moc_dabdemod.cpp:10,
                 from /opt/build/sdrangel/build/plugins/channelrx/demoddab/demoddab_autogen/mocs_compilation.cpp:2:
/opt/build/sdrangel/plugins/channelrx/demoddab/dabdemodsink.h:118:5: error: ‘API_struct’ does not name a type; did you mean ‘struct’?
  118 |     API_struct m_api;
      |     ^~~~~~~~~~
      |     struct
make[2]: *** [plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/build.make:63 : plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/demoddab_autogen/mocs_compilation.cpp.o] Erreur 1
make[2] : on quitte le répertoire « /opt/build/sdrangel/build »
make[1]: *** [CMakeFiles/Makefile2:5823 : plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/all] Erreur 2
make[1] : on quitte le répertoire « /opt/build/sdrangel/build »
make: *** [Makefile:152 : all] Erreur 2
tutu@tutu-Aspire-E5-573:/opt/build/sdrangel/build$

srcejon, 2021-05-05:
You need to update the DAB library to use: https://github.com/srcejon/dab-cmdline

choupette-change, 2021-05-05:
I did it it's the same error ...

[ 36%] Building CXX object plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/demoddab_autogen/mocs_compilation.cpp.o
In file included from /opt/build/sdrangel/plugins/channelrx/demoddab/dabdemodbaseband.h:29,
                 from /opt/build/sdrangel/plugins/channelrx/demoddab/dabdemod.h:31,
                 from /opt/build/sdrangel/build/plugins/channelrx/demoddab/demoddab_autogen/EWIEGA46WW/moc_dabdemod.cpp:10,
                 from /opt/build/sdrangel/build/plugins/channelrx/demoddab/demoddab_autogen/mocs_compilation.cpp:2:
/opt/build/sdrangel/plugins/channelrx/demoddab/dabdemodsink.h:118:5: error: ‘API_struct’ does not name a type; did you mean ‘struct’?
  118 |     API_struct m_api;
      |     ^~~~~~~~~~
      |     struct
make[2]: *** [plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/build.make:63 : plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/demoddab_autogen/mocs_compilation.cpp.o] Erreur 1
make[2] : on quitte le répertoire « /opt/build/sdrangel/build »
make[1]: *** [CMakeFiles/Makefile2:5823 : plugins/channelrx/demoddab/CMakeFiles/demoddab.dir/all] Erreur 2
make[1] : on quitte le répertoire « /opt/build/sdrangel/build »
make: *** [Makefile:152 : all] Erreur 2
tutu@tutu-Aspire-E5-573:/opt/build/sdrangel/build$ 

but in cmake I have :
-- Checking for module 'sgp4'
--   No package 'sgp4' found
-- SGP4 LIBRARIES /opt/install/sgp4/lib/libsgp4s.so
-- SGP4 INCLUDE DIRS /opt/install/sgp4/include/SGP4
-- Checking for module 'apt'
--   No package 'apt' found
is this normal ? I compile apt and sgp4 without problem.. (ok for the  6 10 2)

srcejon, 2021-05-05:
Did you get the msvc branch?

git checkout msvc

Your error suggests you don't have the latest version or wrong branch

f4exb, 2021-06-12:
Question answered.

closed 2021-06-12 by f4exb