f4exb/sdrangel #890: USRP MIMO support

f4exb, 2021-05-06:
This will be targeted at the B210 more specifically its Zixhun Wireless Chinese clone but we'll see how to make this as generic as possible. Thus basically this is 2Rx / 2Tx MIMO support. USRP support is already there with input and output plugins and UHD library support so a lot of the work is already done.

closed 2021-05-12 by f4exb

f4exb, 2021-05-13:
Abandoned.

srcejon, 2021-05-17:
Shame, would have liked to have a play with this. What was the problem?

f4exb, 2021-05-17:
I can't get it work with two simultaneous streams. Documentation is vague and examples too trivial. I have just pushed the `feature-890` branch though if you want to have a peek.

f4exb, 2021-05-17:
More specifically it seems this yields no samples: https://github.com/f4exb/sdrangel/blob/feature-890/plugins/samplemimo/usrpmimo/usrpmithread.cpp#L196 Tried various options here: https://github.com/f4exb/sdrangel/blob/feature-890/plugins/samplemimo/usrpmimo/usrpmithread.cpp#L166 but didn't get anything workable.

srcejon, 2021-05-17:
Thanks. Will take a look when I get some time.

srcejon, 2021-06-15:
If I change:

-    stream_cmd.time_spec = uhd::time_spec_t(2.0);
+    stream_cmd.time_spec = m_usrp->get_time_now() + uhd::time_spec_t(1.0);

-    m_usrp->set_time_now(uhd::time_spec_t(0.0));
+    //m_usrp->set_time_now(uhd::time_spec_t(0.0));

Then samples_received == m_bufSamples.

f4exb, 2021-06-15:
Interesting... I'll try that.

f4exb, 2021-06-17:
Progressing... I still see nothing on the spectrum but both leds light continuously and I do not have error messages. Of course I changed the start mode by `uhd::stream_cmd_t::STREAM_MODE_START_CONTINUOUS`

FPSychotic, 2024-07-20:
I got also a B210 Mini, also called BlackSDR , quite nice device. Effectively when I can use two ports simultaneously, in any TX or RX configuration but it will crash in the moment that I start the 3rd SMA port, doesn't matter if RX or TX. External clock looks work fine, but I don't know really how to check it. 
Just offering test anything you need. Mine has not Ethernet but USB3. Maybe has Ethernet but maybe is in the GPIO connector instead in a Ethernet socket.