f4exb/sdrangel #892: Scope - trace offset

srcejon, 2021-05-08:
The trace offset (11) in the Scope doesn't appear to work as I'd expect when the scope is not running. I'd expect to be able to use it to scroll through the waveform.  The horizonal axis labels seems to work, but not the waveform.

For example, here's an initial captured waveform with offset of 0, where I'm just viewing first 50ms of 500ms.

[image]

If I increase the offset by 25ms, I'd expect the waveform to move to the left 25ms, so I can see 25ms-75ms. However, it seems the first 25ms of the waveform is zeroed, with the second half remaining as 25ms-50ms. The horizontal axis labels do show 25ms-75ms though.

[image]

srcejon, 2021-05-09:
It looks like the offset works if set before the waveform is captured - but would be nice to capture, then scroll through the waveform.

f4exb, 2021-05-13:
Yes, will be covered in #898

commit 5aa2710b (2021-06-02): GLScope redesign: when time offset is positive switch automatically to display memory index 01 to preserve correctness of display. Fixes #892
commit 3e9b4a4d (2021-06-02): GLScope redesign: when time offset is positive switch automatically to display memory index 01 to preserve correctness of display. Fixes #892
closed 2021-06-02 by f4exb