f4exb/sdrangel #899: DAB / Swagger crash

gvanem, 2021-05-15:
I built everything from today's tip-of-the master using MSVC-2019 (x86).
(built with `-DUSE_FFTW` and 16-bit RX sample-size).

Using the Soapy-SDRPlay2 interface and Wideband-FM plugin works great. But adding the
DAB-demodulator channel and selecting channel-block 12B, `swagger.dll` immediately crashes
while accessing a NULL-ptr:
```
eax=00000000 ebx=25937d10 ecx=00000000 edx=00000000 esi=15b38398 edi=1836f078
eip=637fde7a esp=0093a84c ebp=0093a850 iopl=0         nv up ei pl nz na po nc
cs=0023  ss=002b  ds=002b  es=002b  fs=0053  gs=002b             efl=00210202
swagger!QList<SWGSDRangel::SWGObject *>::detach+0xa:
637fde7a 8b08            mov     ecx,dword ptr [eax]  ds:002b:00000000=????????
```
The call-stack:
```
swagger!QList<SWGSDRangel::SWGObject *>::detach(void)+0xa
swagger!QList<SWGSDRangel::SWGObject *>::begin(void)+0xf
swagger!SWGSDRangel::setValue(void * value = 0x0409d7e4, class QJsonValue obj = class QJsonValue, class QString type = class QString, class QString complexType = class QString)+0x1032
swagger!SWGSDRangel::SWGSoapySDRInputSettings::fromJsonObject(class QJsonObject * pJson = 0x0093bd9c)+0x7fe
swagger!SWGSDRangel::setValue(void * value = 0x0093cbc0, class QJsonValue obj = class QJsonValue, class QString type = class QString, class QString complexType = class QString)+0xe36
swagger!SWGSDRangel::SWGDeviceSettings::fromJsonObject(class QJsonObject * pJson = 0x18c59858)+0x15db
sdrbase!ChannelWebAPIUtils::setCenterFrequency(unsigned int deviceIndex = 0, double frequencyInHz = 225648000)+0x575
demodDAB!DABDemodGUI::on_channel_currentIndexChanged(int index = 0x1d)+0x12b
demodDAB!DABDemodGUI::qt_static_metacall(class QObject * _o = 0x15b38398, QMetaObject::Call _c = InvokeMetaMethod (0), int _id = 0xa, void ** _a = 0x0093ce48)+0x135
demodDAB!DABDemodGUI::qt_metacall(QMetaObject::Call _c = InvokeMetaMethod (0), int _id = 0xa, void ** _a = 0x0093ce48)+0x4b
Qt5Core!QMetaObject::metacall(class QObject * object = 0x15b38398, QMetaObject::Call cl = InvokeMetaMethod (0), int idx = 0x2c, void ** argv = 0x0093ce48)+0x28
Qt5Core!doActivate<0>(class QObject * sender = 0x25937d10, int signal_index = 0xc, void ** argv = 0x0093ce48)+0x50b
Qt5Core!QMetaObject::activate(class QObject * sender = 0x25937d10, struct QMetaObject * m = 0x642ee4d4, int local_signal_index = 5, void ** argv = 0x0093ce48)+0x42
Qt5Widgets!QComboBox::currentIndexChanged+0x29
Qt5Widgets!QComboBoxPrivate::_q_emitCurrentIndexChanged(class QModelIndex * index = <Value unavailable error>)+0x45
Qt5Widgets!QComboBoxPrivate::setCurrentIndex(class QModelIndex * mi = 0x0093cfec)+0x142
Qt5Widgets!QComboBoxPrivate::_q_itemSelected+0x34
Qt5Widgets!QComboBox::qt_static_metacall(class QObject * _o = 0x25937d10, QMetaObject::Call _c = <Value unavailable error>, int _id = 0xf, void ** _a = 0x0093cfb4)+0x1bf
Qt5Core!doActivate<0>(class QObject * sender = 0x25c39ed8, int signal_index = 7, void ** argv = 0x0093cfb4)+0x4f3
Qt5Core!QMetaObject::activate(class QObject * sender = 0x25c39ed8, struct QMetaObject * m = 0x642ee51c, int local_signal_index = 0, void ** argv = 0x0093cfb4)+0x42
Qt5Widgets!QComboBoxPrivateContainer::itemSelected(class QModelIndex * _t1 = 0x0093cfec)+0x25
Qt5Widgets!QComboBoxPrivateContainer::eventFilter(class QObject * o = 0x25f8aa30, class QEvent * e = 0x0093d3f0)+0xa7
Qt5Core!QCoreApplicationPrivate::sendThroughObjectEventFilters(class QObject * receiver = 0x25f8aa30, class QEvent * event = 0x0093d3f0)+0xa6
Qt5Widgets!QApplicationPrivate::notify_helper(class QObject * receiver = 0x25f8aa30, class QEvent * e = 0x0093d3f0)+0xb3
Qt5Widgets!QApplication::notify(class QObject * receiver = 0x25f8aa30, class QEvent * e = 0x0093d3f0)+0x6d8
Qt5Core!QCoreApplication::notifyInternal2(class QObject * receiver = 0x25f8aa30, class QEvent * event = 0x0093d3f0)+0xc2
Qt5Widgets!QApplicationPrivate::sendMouseEvent(class QWidget * receiver = 0x25f8aa30, class QMouseEvent * event = 0x0093d3f0, class QWidget * alienWidget = <Value unavailable error>, class QWidget * nativeWidget = <Value unavailable error>, class QWidget ** buttonDown = <Value unavailable error>, class QPointer<QWidget> * lastMouseReceiver = 0x644351d0, bool spontaneous = true, bool onlyDispatchEnterLeave = false)+0x315
Qt5Widgets!QWidgetWindow::handleMouseEvent(class QMouseEvent * event = 0x0093d838)+0x525
Qt5Widgets!QWidgetWindow::event(class QEvent * event = 0x0093d838)+0x2ae
Qt5Widgets!QApplicationPrivate::notify_helper(class QObject * receiver = 0x186763f8, class QEvent * e = 0x0093d838)+0xcd
Qt5Widgets!QApplication::notify(class QObject * receiver = 0x186763f8, class QEvent * e = 0x0093d838)+0x183e
Qt5Core!QCoreApplication::notifyInternal2(class QObject * receiver = 0x186763f8, class QEvent * event = 0x0093d838)+0xc2
Qt5Gui!QGuiApplicationPrivate::processMouseEvent(class QWindowSystemInterfacePrivate::MouseEvent * e = 0x18b273c0)+0x76d
Qt5Gui!QGuiApplicationPrivate::processWindowSystemEvent(class QWindowSystemInterfacePrivate::WindowSystemEvent * e = 0x18b273c0)+0x133
Qt5Gui!QWindowSystemInterface::sendWindowSystemEvents(class QFlags<enum QEventLoop::ProcessEventsFlag> flags = class QFlags<enum QEventLoop::ProcessEventsFlag>)+0x75
qwindows!QWindowsGuiEventDispatcher::sendPostedEvents(void)+0x12
Qt5Core!QEventDispatcherWin32::processEvents(class QFlags<enum QEventLoop::ProcessEventsFlag> flags = class QFlags<enum QEventLoop::ProcessEventsFlag>)+0x67
qwindows!QWindowsGuiEventDispatcher::processEvents(class QFlags<enum QEventLoop::ProcessEventsFlag> flags = class QFlags<enum QEventLoop::ProcessEventsFlag>)+0x15
Qt5Core!QEventLoop::processEvents+0x1c
Qt5Core!QEventLoop::exec(class QFlags<enum QEventLoop::ProcessEventsFlag> flags = class QFlags<enum QEventLoop::ProcessEventsFlag>)+0x1a3
Qt5Core!QCoreApplication::exec(void)+0x144
sdrangel!runQtApplication(int argc = 1, char ** argv = 0x00bfd828, class qtwebapp::LoggerWithFile * logger = 0x00bf8010)+0x788
sdrangel!main(int argc = 1, char ** argv = 0x00bfd828)+0x94
sdrangel!__scrt_common_main_seh(void)+0xfa
KERNEL32!BaseThreadInitThunk+0x19
ntdll!__RtlUserThreadStart+0x2f
ntdll!_RtlUserThreadStart+0x1b
```

I've no idea which .DLL is causing this.

srcejon, 2021-05-15:
Problem seems to be in ChannelWebAPIUtils::setCenterFrequency when using a Soapy input device.

As a workaround, you can either set the device center frequency manually - or use SdrPlayV3 instead of Soapy.

srcejon, 2021-05-15:
Seems to be from this call:

deviceSettingsResponse.fromJsonObject(*jsonObj);

Where *jsonObj is a valid object containing soapySDRInputSettings. 

soapySDRInputSettings seems to have a more complicated structure than the other device settings, in that it has nested objects.

QJsonObject({"deviceHwType":"SoapySDR","direction":0,"soapySDRInputSettings":{"LOppmTenths":0,"antenna":"NONE","autoDCCorrection":0,"autoGain":0,"autoIQCorrection":0,"bandwidth":1000000,"centerFrequency":162000000,"dcCorrection":{"imag":0,"real":0},"devSampleRate":4000,"deviceArgSettings":[{"key":"sample_offset","valueString":"0","valueType":"string"}],"fcPos":2,"globalGain":0,"iqCorrection":{"imag":0,"real":0},"iqOrder":1,"log2Decim":0,"reverseAPIAddress":"127.0.0.1","reverseAPIDeviceIndex":0,"reverseAPIPort":8888,"softDCCorrection":0,"softIQCorrection":0,"streamArgSettings":[{"key":"chan","valueString":"mono_l","valueType":"string"}],"transverterDeltaFrequency":0,"transverterMode":0,"useReverseAPI":0}})

gvanem, 2021-05-15:
>  or use SdrPlayV3 instead of Soapy.

This simply  says `"Could not start sample source"` w/o any explanation.<br>
And i's dropping samples like hell:
```
SampleSinkFifo::write: 119 messages dropped
SampleSinkFifo::write: overflow - dropping 8192 samples
SampleSinkFifo::write: overflow - dropping 8192 samples
```

Welle-IO works fine with the same hardware.

srcejon, 2021-05-15:
What HW are you using?

gvanem, 2021-05-15:
SDRPlay RSP1A and a cheap RTLSDR stick. Both have separate antennas.

f4exb, 2021-05-15:
I have replayed a similar scenario with the swagger UI but with a LimeSDR behind Soapy with no issues. The settings structure is at least as complex. First get the device settings and copy them in the clipboard then patch pasting the clipboard (eliminating the extra "{}" at the beginning) and change the "centerFrequency" value then "execute" the API returns 200 with the new content and the frequency is effectively changed. So in principle this should work.

gvanem, 2021-05-15:
Same as I see. 

> starting from DAB GUI

what is this?

f4exb, 2021-05-15:
Would "down from DAB GUI" make more sense to you?

f4exb, 2021-05-15:
```
#6  SWGSDRangel::setValue (value=0x55ce5a6fea00, obj=..., type=..., complexType=...) at /opt/build/sdrangel/swagger/sdrangel/code/qt5/client/SWGHelpers.cpp:145
145                 for (auto item : **output) {
(gdb) p output
$1 = (QList<SWGSDRangel::SWGObject*> **) 0x55ce5a6fea00
(gdb) p *output
$2 = (QList<SWGSDRangel::SWGObject*> *) 0x0
(gdb) 
```

f4exb, 2021-06-13:
It takes one step back in the stack trace to better spot the issue:
```
#7  0x00007f49adbe975a in SWGSDRangel::SWGSoapySDRInputSettings::fromJsonObject (this=0x557c389dc8b0, pJson=...) at /opt/build/sdrangel/swagger/sdrangel/code/qt5/client/SWGSoapySDRInputSettings.cpp:243
243         ::SWGSDRangel::setValue(&tunable_elements, pJson["tunableElements"], "QList", "SWGArgValue");
(gdb) p tunable_elements
$1 = (QList<SWGSDRangel::SWGArgValue*> *) 0x0
```

f4exb, 2021-06-13:
Issue seems to be with the way `init()` is implemented across objects in the Swagger generator. 
`SWGSDRangel::SWGDeviceSettings::init()` is called but in the implemetation a "new" with the default constructor is called on every possible devices settings:
```c++
void
SWGDeviceSettings::init() {
    device_hw_type = new QString("");
    m_device_hw_type_isSet = false;
    direction = 0;
    m_direction_isSet = false;
    originator_index = 0;
    m_originator_index_isSet = false;
    airspy_settings = new SWGAirspySettings();
    m_airspy_settings_isSet = false;
...
    soapy_sdr_input_settings = new SWGSoapySDRInputSettings();
    m_soapy_sdr_input_settings_isSet = false;
...
}
```
However this default constructor in turn does not invoke its `init()` method instead it sets `nullptr` for every sub object:
```c++
void
SWGSoapySDRInputSettings::SWGSoapySDRInputSettings() {
    center_frequency = 0L;
    m_center_frequency_isSet = false;
...
    tunable_elements = nullptr;
    m_tunable_elements_isSet = false;
...
}
```
Conversely its `init()` method creates an empty list for the `tunable_elements`:
```c++
void
SWGSoapySDRInputSettings::init() {
    center_frequency = 0L;
    m_center_frequency_isSet = false; 
...
    tunable_elements = new QList<SWGArgValue*>();
    m_tunable_elements_isSet = false;
...
}
```
In the constructor with JSON parameter the `init()` method is called:
```c++
SWGSoapySDRInputSettings::SWGSoapySDRInputSettings(QString* json) {
    init();
    this->fromJson(*json);
}
```
So possibly the `init()` method should be called in the default constructor instead of initializing sub objects with `nullptr`.

f4exb, 2021-06-14:
Alternatively with less possible side effects the generated `SWGModelFactory` could be modified this way:
```c++
    if(QString("SWGSoapySDRInputSettings").compare(type) == 0) {
      SWGSoapySDRInputSettings *obj = new SWGSoapySDRInputSettings();
      obj->init();
      return obj;
    }
    if(QString("SWGSoapySDROutputSettings").compare(type) == 0) {
      return new SWGSoapySDROutputSettings();
    }
```
See process for `SWGSoapySDRInputSettings` vs `SWGSoapySDROutputSettings`

f4exb, 2021-06-14:
This refers to my very specific fork of swagger-codegen: https://github.com/f4exb/swagger-codegen/tree/sdrangel

gvanem, 2021-06-14:
This does not make any sense to me.

But if Swagger is to blame, is there a way to build and use SDRangel without it?
Or is this web-sever stuff such an integral part of it? I certainly don't need it for anything.

f4exb, 2021-06-14:
If you don't like SDRangel as it is please either don't use it or fork your own!

f4exb, 2021-06-14:
0-

closed 2021-06-14 by f4exb

gvanem, 2021-06-14:
> If you don't like SDRangel as it is please either don't use it or fork your own! 

I've not stated *I do not like SDRangel*. So this is not an attitude suitable for Open Source projects; 
it won't get much *cred* or good-will from other potential contributors.

f4exb, 2021-06-14:
> This does not make any sense to me.

Is the phrase that really upsets me. What I have tried to develop in this issue which is a real issue is to find the root cause of the problem and gather all elements in order to fix it. So you can't just brush it aside. This is not respectful and offensive for myself and other contributors (not just potential).

gvanem, 2021-06-15:
> So you can't just brush it aside. This is not respectful and offensive for myself ..

A total misunderstanding on your part. Since I do not know C++ that well (and certainly not Swagger/Java), I stated that "*this does not make any sense to me*".

commit a59a58f6 (2021-06-14): Fixed SWGModelFactory so that init() is invoked on complex objects. Fixes #899