f4exb/sdrangel #902: Preset does not restore TX device and sample device

eggelene, 2021-05-16:
System: Ubuntu 20.04
SDRangel 6.12.1
Connected to USB: 2x LimeSDR

How to reproduce:
remove all sink devices and restart SDRAngel ( to restore default settings)
Add sink device ( T1 is added ); configure T1 sink ( select HW and set all parameters)
Add channel ( File source ) ;  select file, enable repeat&play
Save preset
close SDRAngel
open SDRAngel

observation: T1 is missing ( R0 is restored by default )
By loading preset, T1 with sample device is not created.

observation: After manually creating T1, sample device is not restored from preset

observation: After T1 and the sample device are manually created, the right settings and channel is loaded from preset. However the filename for file source is not restored. 

Expected behavior:   sink+ sample device+settings from file source is restored from preset. This is very useful when having multiple file-sources with different settings.

f4exb, 2021-05-16:
response 1: a preset stores settings for just one device set and a set of channels. This ensures enough granularity. In no way a preset will store data for many device sets: R0, T1, R2, ... etc... this is by design and will not change

response 2: at shutdown / startup time only the first device set (R0) is saved as a default preset and restored. This will not change either.

Quetions answered.

closed 2021-05-16 by f4exb

eggelene, 2021-05-16:
thanks for the response.

**_response 1: a preset stores settings for just one device set and a set of channels._** 
Question:  When manually creating sink and sample device, then restoring preset, is it intended that the the file previously selected in the file source ( from channel) is not loaded from preset / saved to preset ? The selected gain from file source is restored though. To me, restoring the filename would be really helpful.

**_response 2: at shutdown / startup time only the first device set (R0) is saved as a default preset and restored. This will not change either._**
Question: so it is by design that the user always needs to manually create sink (T1)  and select sample device after start SDRAngel when using transmit functionality ?

f4exb, 2021-05-17:
>  so it is by design that the user always needs to manually create sink (T1) and select sample device after start SDRAngel when using transmit functionality ?

yes definitely. However you can configure an SDRangel instance pretty much the way you like with the [config.py](https://github.com/f4exb/sdrangel/tree/master/scriptsapi#configpy) script. The JSON file has to be composed manually and maybe I could write another script that dumps the configuration of a SDRangel instance to a similar JSON file thus closing the loop. 

Regarding the various file related plugins maybe not all of them keep the file name persistent and this could be made possible. If you go through the REST API with `config.py` you can specify the file name.

f4exb, 2021-05-17:
7500932c7 fixes FileOutput file name persistence and some REST API missing bits.

eggelene, 2021-05-22:
Thanks a lot for the fixes in the FileOutput. 

I looked into how to manually create the JSON for the config.py script. Maybe I did not search long enough, but could not figure what the structure for the JSON needs to be ( name of the settings and so). Do you have a link for so for me where I can find that information ? Are you planning to make a script that dumps the existing SDRangle configuration ?

N5BRG, 2021-05-22:
If you look on the GitHub at:

https://github.com/f4exb/sdrangel/tree/master/swagger/sdrangel/examples

You will see examples of Python code using the json. This should be enough to help you do what you need.

I have been working on an application for Hamlib to also this package to interface with SDRangel. Currently the code uses CURL to do the REST interface. In an effort to speed this up I am converting the CURL to python.
I have had side delays recently but hope to finish this soon.

There is also a Feature RIGCTL interface in SDRangel but I have not seen that work yet.

With the Hamlib RIGCTL interface you can send commands for a script like:

F 144000000   Sets the freq
f Read the freq
T 1 Move to Tx and activate Tx
T 0 Move to Rx and activate Rx

These can be done by hand or from a script.

My code does work to do this now. It is not fast enough to support the WSJTX modes which were important to me. I have been working to speed it up but the REST interface may prove to be too slow.

You can find a linux version of my effort at:

https://github.com/N5BRG/Hamlib

Bob N5BRG