f4exb/sdrangel #911: gs232controller- Add "tolerance" parameter ?

Valstee, 2021-05-28:
Hi F4exb,
Is it possible to add a tolerance parameter (**y**) to avoid jerks when the motor executes a command? If the setpoint is **x** degrees, the response can be **x** degrees +/- **y** degrees. Because the controller gives micro "instructions" as long as the engine is not perfectly at x degree.
I'm not competent enough to add this function...
Thanks a lot,
Valentin

srcejon, 2021-05-28:
Is this when used with the Satellite Tracker or another feature? The GS232 Controller itself only sends one set command in response a change in Az/El in the GUI.

If it is the Sat Tracker, you could try reducing the update period under Settings > Display.

But yes, this can be added.

Valstee, 2021-05-31:
Hi @srcejon, 
It happens when I use sat tracker but I think I also have some mechanical adjustments to make... I changed the update period T but I still have small movements every T periodes around the Az/El command.
Thanks a lot for adding this feature !
Valentin

closed 2021-05-31 by Valstee