f4exb/sdrangel #920: ERRO  HACKERF ONE

gaspar1987, 2021-06-13:
Hello Hello when I connect with the HACKERF ONE key The software does not work and it breaks when I connect and I would like to know that it comes from your software which is faulty if you can establish a better preparation to repair it I have to put two keys instead of one it is a problem that I work on several I work on several hardware and I do not make the hardware that I want and that's why I want you to fix this problem of the keys of all the keys that is compatible with your software because I had the bad surprise that the software breaks when I plug in Emma my key it breaks during startup when I press PLAY that's it but it breaks thank you for fixing the problem I find that your software is very professional and I approve of an exceptional quality cordially good day Sir Thank you to YOU distinguished program

gaspar1987, 2021-06-13:
[image]

gaspar1987, 2021-06-13:
Quand j'appuie ok ça ne fonctionne pas j'aimerais savoir quel problème que vous faites pour résoudre ce problème là merci et à bientôt

f4exb, 2021-06-13:
Are you running on Windows? It is a known story that HackRF support broke at some point probably for some reason buried deep into `libusb`: https://github.com/f4exb/sdrangel/issues/718

People seem to have success in working around the issue by using HackRF through SoapySDR.

gaspar1987, 2021-06-14:
D'habitude ça marche on vous ancien logiciel je comprends pas pourquoi ça ne fonctionne pas sur les nouveaux..

gaspar1987, 2021-06-14:
Essayer de réparer le logiciel en lui mettant des répartitions qui permet de réparer avec la clé et essayer de tester vous-même vous verrez que vos logiciel tombe en panne avec le hacker f one

f4exb, 2021-06-14:
Closed in favor of #718

closed 2021-06-14 by f4exb