f4exb/sdrangel #924: Windows version 6 14 0 doesn't start

choupette-change, 2021-06-15:
the windows version 6 14 0 does not start ... the launch process is interrupted ???
[image]

srcejon, 2021-06-15:
What's in the log file?

choupette-change, 2021-06-15:
i haven't changed anything from the previous version, i just uninstalled it and reinstalled 6 14 0, where is the log file?

srcejon, 2021-06-15:
See Preferences > Logging menu

Usually in c:\Users\you\sdrangel.log

choupette-change, 2021-06-15:
i can't open V6 14 0 so i have no log file. I just retested the V 6 13 it's ok for me ...

f4exb, 2021-06-16:
If you set up the log file in v6.13 these settings will be persistent anyway when you upgrade to v6.14. As the issue is not trivially reproducible we need more information to investigate the issue further.

choupette-change, 2021-06-16:
[sdrangel.log](https://github.com/f4exb/sdrangel/files/6662424/sdrangel.log)

f4exb, 2021-06-16:
This is the log from the 6.13.0 version and I do not understand why `DeviceAPI::loadSamplingDeviceSettings` call happens 3 times. However I see the AIS demod is in the default preset therefore it gets loaded at startup time. There is an issue in 6.14.0 with AIS demod already reported in #925 and this may very well be the issue here.

Can you try with the latest version from master. There is an installer directly accessible from Appveyor here: https://ci.appveyor.com/project/f4exb/sdrangel/builds/39618839/job/9r81isna2rm3b6ms/artifacts

choupette-change, 2021-06-16:
it works with the link above (master); I attach you the last log file created
[sdrangel.log](https://github.com/f4exb/sdrangel/files/6662936/sdrangel.log)

f4exb, 2021-06-23:
ok then it is fixed with #925

closed 2021-06-23 by f4exb