f4exb/sdrangel #925: Segmentation error caused by AIS Demodulator

Valstee, 2021-06-15:
Hi, I use sdrangel 6.14 on Ubuntu 20.10.
The opening of Ais demodulator causes a segmentation error.
Here are the logs :
- (W) QMetaObject::connectSlotsByName: No matching signal for on_scope_sampleRateChanged(int)
- (W) QMetaObject::connectSlotsByName: No matching signal for on_scope_traceSizeChanged(uint32_t)
- (W) QMetaObject::connectSlotsByName: No matching signal for on_scope_preTriggerChanged(uint32_t)
Do you have an idea of the problem?
Thanks,
Valentin

choupette-change, 2021-06-15:
I have the same issue with Ubuntu 21 04 with AIS demodulator. 
Other issue with plugin : Noise figure (cf print screen)
[image]

f4exb, 2021-06-15:
@choupette-change no AIS demodulator here...

f4exb, 2021-06-15:
@choupette-change please open another issue for the Noise Figure and let's keep this one for the original issue.

srcejon, 2021-06-15:
I see an exception: vector subscript out of range

In: 

void GLScopeGUI::settingsTraceChange(const GLScopeSettings::TraceData& traceData, uint32_t traceIndex)
{
    m_settings.m_tracesData[traceIndex] = traceData;
}

traceIndex is 0.

Stack is:

[image]

Calling line is AISDemod is:

ui->scopeGUI->addTrace(traceDataQ);

f4exb, 2021-06-15:
Yes it is probably due to the recent changes in the scope. It looks like it does not start with one trace already defined.

closed 2021-06-15 by f4exb

f4exb, 2021-06-16:
Corrected again: issue was actually in `GLScopeGUI::setBuddies` and not the plugin(s) that call it. The previous fix results in one extra trace or trigger created. Changes in plugin have been reverted.

commit 07a81679 (2021-06-15): AIS demod GUI: adapt to new scope. Add first trace and first trigger. Fixes #925