f4exb/sdrangel #926: issue with plugin : Noise figure

choupette-change, 2021-06-15:
Other issue with plugin : Noise figure (cf print screen)
[image]

srcejon, 2021-06-15:
The noise figure plugin requires a calibrated noise source? Do you have one? If so, which one?

The noise source needs to have it's power switched on and off. By default, it assumes a 346B type device and so it tries to do this via a programmable power supply connected via USB using the VISA library. This error is saying it can't communicate with the power supply. If you have one, you need to enter the correct VISA address for it in the Control dialog. If you don't, you can use a program or script to control the power supply / enable the noise source instead.

f4exb, 2021-07-01:
Assuming @choupette-change has not the proper equipment.

f4exb, 2021-07-02:
Question answered

closed 2021-07-02 by f4exb