f4exb/sdrangel #929: float feed to scope

srcejon, 2021-06-22:
As I've mentioned before, I'm been using the scope to help debug new modems. It seems the scope can't currently display values smaller than 0.000001 when feed from a channel, even if the vertical range slider is set below 1u. They appear as 0.

This is because the scope interface is using fixed point types, and 8388608*0.0000001 is less than 1. This limit would presumably be .00001 when SDR_RX_SAMP_SZ==16.

This makes sense for the IQ traces, but as a more general purpose tool, it would be useful to feed floats as well, for increased range. It seems ScopeVis::processTraces converts the SampleVector to float for the arithmetic anyway. Could possibly just be done with a template for feed and processTraces.

As another possible extension, the vertical scale could perhaps have a GLScopeSettings to include 5, 10, 100 etc to allow scaling of values > 1, which would be available with float inputs.

f4exb, 2021-06-25:
On latest master...

In `dsp/dsptypes.h`:
```c++
typedef float Real;
typedef std::complex<Real> Complex;
typedef std::vector<Complex> ComplexVector;
```
In `scopevis.h` a new `feed` method takes `ComplexVectgor` as input. In fact this is the method that does the job now since internally samples are `std::complex<float>`:
```c++
void feed(const std::vector<ComplexVector::const_iterator>& vbegin, int nbSamples);
```

The "vertical scale" or amplification factor change to take larger value is to be handled in a new ticket. This is essentially a GUI thing.

srcejon, 2021-06-25:
Thanks!

int ScopeVis::processTraces(const std::vector<ComplexVector::const_iterator uses m_spectrumVis (scopevis.cpp line 763) - which for the Radio Clock plugin is currently nullptr (it only has a scope not a spectrum). Do I need to create one? It didn't seem to be needed before.

[image]

f4exb, 2021-06-25:
This shouldn't be the case. I think it is missing a test of m_spectrumVis nullity. Will be fixed shortly.

srcejon, 2021-06-25:
Shall I push the changes to Radio Clock plugin so you can give it a try?

f4exb, 2021-06-25:
I am really close to push this small change on master so maybe you want to rebase after that.

f4exb, 2021-06-27:
Implemented in v6.15.0

closed 2021-06-27 by f4exb