f4exb/sdrangel #930: Scope: redesign from scratch

f4exb, 2021-06-22:
It seems that the newest features including #929 are a call for a redesign from the ground up. There is nothing to be kept at all.

srcejon, 2021-06-22:
I hope I didn't sound too negative. The scope is really useful as is! I thought it would be just a relatively small change to implement 929 (but I must admit, I only looked at the feed functions - so could have missed lots of complexity).

f4exb, 2021-06-22:
The problem is that the scope components have endured patches over patches for many years making the code very hard to maintain. Re 929 the data type is carried over fairly deep and a scaled signal is assumed not just any value which would make support of scale values larger that +/-1 fairly hard to implement.

f4exb, 2021-06-22:
Maybe the immediate need of 929 can be accommodated without a complete redesign from the ground up but this time I'll keep the issue open because this is probably needed in the long run.

f4exb, 2021-12-02:
The overall design is not great but it works and 929 is implemented. So there are probably better things to do ...

closed 2021-12-02 by f4exb