f4exb/sdrangel #940: ATV Crash when adding demodulator

MMSSolutions, 2021-06-25:
Version 6.14.0

Currently have an issue where I go to add an ATV demodulator, but SDRAngel crashes immediately. This is a fresh install on a fresh install of windows.  Other demodulators work fine.

Any recommendations to fix or is this a current bug?

I have also tried to run as admin, but no luck, still crashes.

I have tried Pluto and Airspy devices.

Thanks
Matt

srcejon, 2021-06-25:
Quite possibly related to the Scope issue: https://github.com/f4exb/sdrangel/issues/925 that has been fixed, as the ATV includes the scope.

f4exb, 2021-06-26:
You can try the installer that you will find here: https://ci.appveyor.com/project/f4exb/sdrangel/builds/39754826/job/6phchv69ogl72ocb/artifacts

MMSSolutions, 2021-06-26:
Thanks! That works fine.

f4exb, 2021-06-27:
Fix deployed in v6.15.0

closed 2021-06-27 by f4exb