f4exb/sdrangel #943: Not possible to compile in linux

mahoo4you, 2021-06-29:
[ 40%] Building CXX object pluginssrv/channelrx/demodais/CMakeFiles/demodaissrv.dir/demodaissrv_autogen/mocs_compilation.cpp.o
/home/mahoo/sdr/opt/build/sdrangel/plugins/channelrx/noisefigure/noisefigure.cpp: In member function ‘double NoiseFigure::calcENR(double)’:
/home/mahoo/sdr/opt/build/sdrangel/plugins/channelrx/noisefigure/noisefigure.cpp:183:100: error: no matching function for call to ‘boost::math::barycentric_rational<double>::barycentric_rational(std::remove_reference<std::vector<double>&>::type, std::remove_reference<std::vector<double>&>::type, int&)’
             boost::math::barycentric_rational<double> interpolant(std::move(x), std::move(y), order);
                                                                                                    ^
In file included from /home/mahoo/sdr/opt/build/sdrangel/plugins/channelrx/noisefigure/noisefigure.cpp:19:
/usr/include/boost/math/interpolators/barycentric_rational.hpp:40:5: note: candidate: ‘template<class InputIterator1, class InputIterator2> boost::math::barycentric_rational<Real>::barycentric_rational(InputIterator1, InputIterator1, InputIterator2, size_t, typename boost::disable_if_c<boost::is_integral<InputIterator2>::value>::type*)’
     barycentric_rational(InputIterator1 start_x, InputIterator1 end_x, InputIterator2 start_y, size_t approximation_order = 3, typename boost::disable_if_c<boost::is_integral<InputIterator2>::value>::type* = 0);
     ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/math/interpolators/barycentric_rational.hpp:40:5: note:   template argument deduction/substitution failed:
In file included from /home/mahoo/sdr/opt/build/sdrangel/plugins/channelrx/noisefigure/noisefigure.cpp:19:
/usr/include/boost/math/interpolators/barycentric_rational.hpp: In substitution of ‘template<class InputIterator1, class InputIterator2> boost::math::barycentric_rational<double>::barycentric_rational(InputIterator1, InputIterator1, InputIterator2, size_t, typename boost::disable_if_c<boost::is_integral<InputIterator2>::value, void>::type*) [with InputIterator1 = std::vector<double>; InputIterator2 = int]’:
/home/mahoo/sdr/opt/build/sdrangel/plugins/channelrx/noisefigure/noisefigure.cpp:183:100:   required from here
/usr/include/boost/math/interpolators/barycentric_rational.hpp:57:1: error: no type named ‘type’ in ‘struct boost::disable_if_c<true, void>’
 barycentric_rational<Real>::barycentric_rational(InputIterator1 start_x, InputIterator1 end_x, InputIterator2 start_y, size_t approximation_order, typename boost::disable_if_c<boost::is_integral<InputIterator2>::value>::type*)
 ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/boost/math/interpolators/barycentric_rational.hpp:49:1: note: candidate: ‘boost::math::barycentric_rational<Real>::barycentric_rational(const Real*, const Real*, size_t, size_t) [with Real = double; size_t = long unsigned int]’
 barycentric_rational<Real>::barycentric_rational(const Real* const x, const Real* const y, size_t n, size_t approximation_order):
 ^~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/boost/math/interpolators/barycentric_rational.hpp:49:1: note:   no known conversion for argument 1 from ‘std::remove_reference<std::vector<double>&>::type’ {aka ‘std::vector<double>’} to ‘const double*’
In file included from /home/mahoo/sdr/opt/build/sdrangel/plugins/channelrx/noisefigure/noisefigure.cpp:19:
/usr/include/boost/math/interpolators/barycentric_rational.hpp:34:7: note: candidate: ‘boost::math::barycentric_rational<double>::barycentric_rational(const boost::math::barycentric_rational<double>&)’
 class barycentric_rational
       ^~~~~~~~~~~~~~~~~~~~
/usr/include/boost/math/interpolators/barycentric_rational.hpp:34:7: note:   candidate expects 1 argument, 3 provided
/usr/include/boost/math/interpolators/barycentric_rational.hpp:34:7: note: candidate: ‘boost::math::barycentric_rational<double>::barycentric_rational(boost::math::barycentric_rational<double>&&)’
/usr/include/boost/math/interpolators/barycentric_rational.hpp:34:7: note:   candidate expects 1 argument, 3 provided
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/transverterdialog.cpp.o
make[2]: *** [pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/build.make:63: pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/noisefigure.cpp.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:10456: pluginssrv/channelrx/noisefigure/CMakeFiles/noisefiguresrv.dir/all] Fehler 2
make[1]: *** Es wird auf noch nicht beendete Prozesse gewartet....
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/tvscreen.cpp.o
[ 40%] Building CXX object pluginssrv/channelrx/radioclock/CMakeFiles/radioclocksrv.dir/radioclocksettings.cpp.o
[ 40%] Linking CXX shared library ../../../lib/pluginssrv/libdemodaissrv.so
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/tvscreenanalog.cpp.o
[ 40%] Building CXX object pluginssrv/channelrx/radioclock/CMakeFiles/radioclocksrv.dir/radioclockbaseband.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/valuedial.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/valuedialz.cpp.o
[ 40%] Building CXX object pluginssrv/channelrx/radioclock/CMakeFiles/radioclocksrv.dir/radioclocksink.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/wsspectrumsettingsdialog.cpp.o
[ 40%] Built target demodaissrv
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/gui/wrappingdatetimeedit.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/dsp/scopevisxy.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/device/deviceuiset.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/channel/channelgui.cpp.o
[ 40%] Building CXX object pluginssrv/channelrx/radioclock/CMakeFiles/radioclocksrv.dir/radioclockplugin.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/feature/featuregui.cpp.o
[ 40%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/feature/featureuiset.cpp.o
[ 40%] Building CXX object pluginssrv/channelrx/radioclock/CMakeFiles/radioclocksrv.dir/radioclockwebapiadapter.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/discreterangegui.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/intervalrangegui.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/itemsettinggui.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/stringrangegui.cpp.o
[ 41%] Building CXX object pluginssrv/channelrx/radioclock/CMakeFiles/radioclocksrv.dir/radioclocksrv_autogen/mocs_compilation.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/dynamicitemsettinggui.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/dynamicargsettinggui.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/intervalslidergui.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/complexfactorgui.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/soapygui/arginfogui.cpp.o
[ 41%] Linking CXX shared library ../../../lib/pluginssrv/libradioclocksrv.so
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/sdrgui_autogen/mocs_compilation.cpp.o
[ 41%] Building CXX object sdrgui/CMakeFiles/sdrgui.dir/sdrgui_autogen/3YJK5W5UP7/qrc_res.cpp.o
[ 41%] Built target radioclocksrv
[ 41%] Linking CXX shared library ../lib/libsdrgui.so
[ 41%] Built target sdrgui
make: *** [Makefile:152: all] Fehler 2

srcejon, 2021-06-30:
What version if boost do you have installed?

mahoo4you, 2021-06-30:
i am running debian with libboost-dev 1.67.0.1

srcejon, 2021-06-30:
Looks like that constructor was introduced in boost 1.70.0

https://www.boost.org/doc/libs/1_70_0/libs/math/doc/html/math_toolkit/barycentric.html

Can you upgrade boost? If not, you could just disable the noise figure plugin as a workaround.

srcejon, 2021-06-30:
@f4exb - do you want older versions of boost supported? I could probably rewrite that code.

mahoo4you, 2021-06-30:
@srcejon i will try :) tnx guys ! 73 de mat

f4exb, 2021-06-30:
@srcejon I don't think this is necessary. Appveyor builds work, Docker works and of course in the recommended Ubuntu 21.04 disto.

mahoo4you, 2021-06-30:
@srcejon it works with the disabled noise figure plugin. tnx guys for sdrangel . really a great software. 73 de mat

commit af1dc386 (2021-07-01): Compile Noise Figure plugin only if Boost version is greater or equal to 1.70.0. Fixes #943
closed 2021-07-01 by f4exb