f4exb/sdrangel #947: Sdrangel display freezes

choupette-change, 2021-07-02:
After about 10 to 15 minutes of operation, the sdrangel display freezes ??? Is this a problem with my video card, your opinion please? Especially if I have several plugins or features that work simultaneously .. SDRangel no longer responds
(Windows and Ubuntu version). Thank you for your feedback
[image]

f4exb, 2021-07-02:
Impossible to say anything with so little information

choupette-change, 2021-07-03:
[sdrangel.log](https://github.com/f4exb/sdrangel/files/6758399/sdrangel.log)

f4exb, 2021-07-03:
Apparently you are loading a preset called `ais_rtl`. I'd like to know what's inside. You can start SDRangel without starting anything and from another terminal run: `curl -X GET "http://localhost:8091/sdrangel/config"`. This should return a possibly big (depends on how many presets you have) JSON output. If you can post that JSON this could be helpful as it contains the details of the `ais_rtl` preset. 

I can guess by the name and some hints in the log what it should look like but I need to reproduce with the exact same settings. 

From the log it appears that when you start the RTL-SDR some locking occurs as demonstrated by the numerous `SampleSinkFifo::write: overflow` messages. Then the program appears as "frozen".

choupette-change, 2021-07-06:
ais_rtl.prex:
kAABAXABAHACB2Fpc19ydGwwAwQJp+yAgQQBlQAAAP8AAAAA/QAAAAMAAAAAAAABUQAAApf8AgAAAAP7AAAAGgBpAG4AcAB1AHQAVgBpAGUAdwBEAG8AYwBrAQAAABMAAAEuAAAA3gAAAS77AAAAJABzAHAAZQBjAHQAcgBhAEQAaQBzAHAAbABhAHkARABvAGMAawEAAAFHAAAAhgAAAIYAAACG/AAAAdMAAADXAAAAkQEAABn6AAAAAAIAAAAC+wAAABQAcAByAGUAcwBlAHQARABvAGMAawEAAAAA/////wAAAHcA////+wAAABgAYwBvAG0AbQBhAG4AZABzAEQAbwBjAGsBAAAAAP////8AAAB3AP///wAAAAEAAAFiAAACl/wCAAAAAfsAAAAWAGYAZQBhAHQAdQByAGUARABvAGMAawEAAAATAAAClwAAAJEA////AAAAAgAAAmkAAAHF/AEAAAAB+wAAABYAYwBoAGEAbgBuAGUAbABEAG8AYwBrAQAAAVcAAAJpAAAAbQD///8AAAJpAAAAzAAAAAEAAAACAAAAAQAAAAL8AAAAAIAFe5AAAQEAAQIEAAACAAADAQRABATB6AAAQAUEQdAAAAAaARRgBgEAYAcBAWAIAQBgCQEAAAoBAWALAQEADQEXAA4BAQAPAR5gEAEBABEBMkASBD8o9cMAEwEBABQCAMhgFQEAcBYJMTI3LjAuMC4xEBcCIrdgGAEAYBkBAWAGAQEABwAAFAEBcBgcc2RyYW5nZWwuc2FtcGxlc291cmNlLnJ0bHNkcnAZCDAwMDAwMDAxABoBAYAbX5AAAQEAAgIB8AADABAEAQVgBQEBYAYBAAAHAQIACAMfQABgCQEAYAoBAWALAQBgDAEAIA0AEA4DAYagYA8BAGAQAQBwEQkxMjcuMC4wLjEQEgIiuBATAGAUAQFgFQEAAMgBAnDJGXNkcmFuZ2VsLmNoYW5uZWwuYWlzZGVtb2SBygFkkAABAQABAp5YQAIERkNQAEADBEWWAABABAQAAAAAcAUAYAYBAHAHCTEyNy4wLjAuMRAIAicPAAkAAAoAAAsAEAwE/1UA/3AND0FJUyBEZW1vZHVsYXRvcoAOJZAAAQEAAQKeWBACBP9VAP9wAw9BSVMgRGVtb2R1bGF0b3IABwAADwBgEAEAcBEJMTI3LjAuMC4xEBICIrgQEwAQFACAFYyQAAEBAAEBAgACAUsAAwEkAAQBCAAFAAAGAQEQCgEBABQAQBUEP4AAAEAWBAAAAAAAGAAAGQBAGgQ/gAAAQBsEP4AAAEAcBD6AgIEQHQAQyAEBAMkBAQDSAADTAGDUAQFg1QEAANYAANcAANgAANkAQNoEAAAAAEDbBD+AAABA3AQAAAAAEN0BARDeAABkAABlAQEAZgECAGcBAwBoAQQAaQEFAGoBBgDIAf8AyQH/AMoB/wDLAf8AzAH/AM0B/wDOAf9wyxlzZHJhbmdlbC5jaGFubmVsLmFpc2RlbW9kgcwBn5AAAQEAAQJhqEACBEZDUABAAwRFlgAAQAQEAAAAAHAFCTIyODA0MDYwMGAGAQBwBwkxMjcuMC4wLjEQCAInDwAJAAAKAQUACwEGEAwE//8AAHAND0FJUyBEZW1vZHVsYXRvcoAOJZAAAQEAAQJhqBACBP//AABwAw9BSVMgRGVtb2R1bGF0b3IABwAADwBgEAEAcBEJMTI3LjAuMC4xEBICIrgQEwAQFACAFbyQAAEBAAEBAgACATIAAwEKAAQBAQAFAAAGAQEQCgECABQAQBUEP4AAAEAWBAAAAAAAGAAAGQBAGgQ/gAAAQBsEP4AAAEAcBD6AgIEQHQAAJAEBQCUEP4AAAEAmBAAAAAAAKAAAKQBAKgQ/gAAAQCsEP4AAAEAsBD6AgIEQLQAQyAEBAMkBAQDSAADTAGDUAQFg1QEAANYAANcAANgAANkAQNoEAAAAAEDbBD+AAABA3AQAAAAAEN0BARDeAABkAABlAQEAZgECAGcBAwBoAQQAaQEFAGoBBgDIAf8AyQH/AMoB/wDLAf8AzAH/AM0B/wDOAf8=

f4exb, 2021-07-07:
Do you also have feature plugins like AIS and Map which would make sense to complete the AIS reception?

f4exb, 2021-07-07:
I can experience some slowness in the GUI when the list of received messages gets long. Maybe the cost of refreshing a table is not linear in Qt but that's Qt business. It can be possibly be mitigated but firstly you should not set the correlation threshold to 0. See here: https://github.com/f4exb/sdrangel/tree/master/plugins/channelrx/demodais#6-th---correlation-threshold

When set to 0 there are a lot of messages with zero MSI and Unknown type that are garbage and fill up the table and also keeps the processors busy for nothing. You have to experiment with this threshold depending on your reception conditions so that "unknown" type messages are kept to a minimum.

Secondly to seek performance a bit more you may change the spectrum settings:
[image]

  - (1) Instead of using a moving average use a fixed average or max
  - (2) limit the FPS to 20 FPS

You can improve efficiency a bit more by choosing a multiple of 57.6 kS/s for the baseband sample rate. You also need enough bandwidth so that is at least 57.6 &times; 2 = 115.2 kS/s. Thus 1843.2 kS/s with a decimation of 16 will do.

choupette-change, 2021-07-07:
Question : "Do you also have feature plugins like AIS and Map which would make sense to complete the AIS reception?"
I use simultaneously with the adsb decoder and AIS with map feature plugins

closed 2021-12-19 by github-actions[bot]