f4exb/sdrangel #948: [Enhancement] Update web API for Satellite Tracker

Valstee, 2021-07-02:
Hi @srcejon, hi @f4exb,
I tried to update the web API to integrate the Satellite Tracker feature. But I didn't succeed.
Is it technically possible? Do you plan to integrate it?
Thanks a lot,
Valentin

f4exb, 2021-07-02:
Can you be more specific?

Valstee, 2021-07-02:
I use the server version (no GUI) on a raspberry to solve some of the problems mentioned in #840 et #845 .
But I can't send PATCH/POST commands for the "Satellite Tracker" feature to set them up. (especially the name of the satellites to be observed and the command to be executed accordingly)
I think it's not implemented yet (i've got a "Error: Invalid JSON request" error 400) . I try to reproduce what @srcejon have done with other feature but I didn't succeed... (for exemple in this commit : https://github.com/f4exb/sdrangel/commit/721d0a40c2ac0751ee5e3d7778bf0ee7684eefb4)
So It's more an enhancement request...

f4exb, 2021-07-02:
In can help if you post the JSON payload of your request

srcejon, 2021-07-02:
The API should be there - E.g: see: https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/api/swagger/include/SatelliteTracker.yaml

But I must admit, I didn't spend much time testing it. The list of satellites to track is an array of strings named "satellites". Perhaps you're just sending a string rather than an array or something like that?

Valstee, 2021-07-02:
What I do to test is: 
- I use sdrangel GUI.
- I open the feature satellite tracker "by hand"
- In the  "Interactive Swagger UI Documentation", I do a GET/settings on the feature and I get this:
`{
  "SatelliteTrackerSettings": {
    "aosSpeech": "${name} is visible for ${duration} minutes. Max elevation, ${elevation} degrees.",
    "autoTarget": 1,
    "azElUnits": 1,
    "dateFormat": "yyyy/MM/dd",
    "defaultFrequency": 100000000,
    "dopplerPeriod": 10,
    "drawOnMap": 1,
    "groundTrackPoints": 100,
    "heightAboveSeaLevel": 500,
    "latitude": 43.2,
    "longitude": 8.9,
    "losSpeech": "${name} is no longer visible.",
    "minAOSElevation": 5,
    "minPassElevation": 15,
    "passFinishTime": "23:59:59",
    "passStartTime": "00:00:00",
    "predictionPeriod": 5,
    "reverseAPIAddress": "127.0.0.1",
    "reverseAPIChannelIndex": 0,
    "reverseAPIDeviceIndex": 0,
    "reverseAPIPort": 8888,
    "rgbColor": -2025117,
    "satellites": [
      "NOAA 15",
      "NOAA 18",
      "NOAA 19"
    ],
    "target": "NOAA 15",
    "title": "Satellite Tracker",
    "tles": [
      "https://db.satnogs.org/api/tle/",
      "https://www.amsat.org/tle/current/nasabare.txt",
      "https://www.celestrak.com/NORAD/elements/goes.txt"
    ],
    "updatePeriod": 1,
    "useReverseAPI": 0,
    "utc": 0
  },
  "featureType": "SatelliteTracker"
}
`
-  I then do a PUT/settings and decide to remove "NOAA18" from the list:
`{
  "SatelliteTrackerSettings": {
    "satellites": [
      "NOAA 15",
      "NOAA 19"
    ]
  },
  "featureType": "SatelliteTracker"
}`
- I've got this answer :
`{
  "message": "Invalid JSON request"
}
`
- And when i look at "example Model" there is nothing so i conclude that is not yet implement ?

[image]

f4exb, 2021-07-02:
Although the PUT should work what you really want to do here is a partial update so rather use PATCH. PUT will take default values for all other settings so may be changing them. However I think it should work in both cases.

srcejon, 2021-07-02:
I see a couple of bugs - looks like I haven't updated some of the code in webapi for the satellite tracker. I'll do that now.

srcejon, 2021-07-02:
The above patch is a partial fix, in that it allows basic settings such as the target to be set. 

However, there is still a problem when trying to set the satellites list. I get a null pointer exception in the swagger code:

[image]

f4exb, 2021-07-02:
Could be an issue with embedded complex objects (lists in that case) as we had before for SoapySDR. I'll have a look.

closed 2021-07-02 by f4exb

f4exb, 2021-07-02:
I must admit this is not very intuitive but if the structure contains embedded objects or lists on has to invoke the `init()` method on it like in the `getChannelSettings()` here: https://github.com/f4exb/sdrangel/blob/master/sdrbase/webapi/webapirequestmapper.cpp#L3868

PR #949 (2021-07-02): Add SatelliteTracker to webapirequestmapper to partially fix #948
Add SatelliteTracker to webapirequestmapper to partially fix #948

This allows basic settings to be set, i.e. target satellite, but it seems there's a null pointer exception when trying to set "satellites" which uses a string array.
commit beeb91f1 (2021-07-02): Add SatelliteTracker to webapirequestmapper to partially fix #948
commit c3c77005 (2021-07-02): Merge pull request #949 from srcejon/fix_948_basic_settings

Add SatelliteTracker to webapirequestmapper to partially fix #948
commit f80b4785 (2021-07-02): Satellite Tracker: invoke init on swagger Satellite Tracker settings since it contains a list. Fixes #948