f4exb/sdrangel #952: SDRPlayV3 plugin: fix for 'sdrplay_api_RfUpdateError' error message

fventuri, 2021-07-05:
@f4exb @srcejon

Edouard, Jon,
In the last few days I have been running the SDRPlayV3 plugin that Jon wrote and I noticed error messages like these when the RSP starts streaming:

> 2021-07-05 16:37:36.454 (C) SDRPlayV3Input::applySettings: could not set bandwidth:  sdrplay_api_RfUpdateError
2021-07-05 16:37:36.859 (C) SDRPlayV3Input::applySettings: could not set IF frequency:  sdrplay_api_RfUpdateError

I think these errors from the SDRplay API are due to the fact that the update calls (`sdrplay_api_Update`) are asynchronous (i.e. the function returns before the actual change is completed). This comment has a more detailed description: https://github.com/fventuri/gr-sdrplay3/issues/5#issuecomment-866848230

This afternoon I wrote a few lines of code for the SDRPlayV3 plugin to make it wait for the RF update to be completed before running the next update in the method `applySettings`, in a similar way to what I did here: https://github.com/pothosware/SoapySDRPlay3/commit/7ffab93ac31438b54cfb27bd9515b3dc251c25d9 a few weeks ago.
I just pushed the changes to the branch `fix_RfUpdateErrors_in_sdrplayv3` here: https://github.com/fventuri/sdrangel/tree/fix_RfUpdateErrors_in_sdrplayv3 - the changes are in this commit: https://github.com/fventuri/sdrangel/commit/be2dbabafee76a199fdfe3b870f703d5c1795282

I ran SDRangel again with these fixes and I don't see those error messages any more.

If you are interested, I can create a pull request, or you can just pick the few changes I made from the commit above and integrate them in the SDRPlayV3 plugin.

Regards,
Franco Venturi, K4VZ

f4exb, 2021-07-06:
The issue must be hardware or installation specific because I never encountered it. It seems by the comment you mention that it is intermittent which is an indication of this situation. But why not preventing a potential issue. However as you might see in my comment in your changes `m_sdrPlayThread` can be `nullptr` if the device is stopped or before it is even started thus you must take this situation into account. While the `m_sdrPlayThread` is not created it is still possible to change parameters.

fventuri, 2021-07-06:
@f4exb - Edouard, good point about being hardware or installation specific - in this case, it probably makes more sense to wait to see if someone else reports a similar problem before adding these changes.

As per `m_sdrPlayThread` possibly being `nullptr`, please notice that the code changes only reference it under the condition `m_running == true` (see here for instance: https://github.com/fventuri/sdrangel/blob/fix_RfUpdateErrors_in_sdrplayv3/plugins/samplesource/sdrplayv3/sdrplayv3input.cpp#L461), and `m_running` is set to the value `true` only after `m_sdrPlayThread` is constructed in line 144, therefore I assumed a check on `m_sdrPlayThread` was redundant.

Franco

f4exb, 2021-07-06:
> As per `m_sdrPlayThread` possibly being `nullptr`, please notice that the code changes only reference it under the condition `m_running == true` 

Yes, just missed that.

PR #954 (2021-07-06): Fix rf update errors in sdrplayv3
The error is intermittent but this is preventive maintenance. Fixes #952
closed 2021-07-06 by f4exb