f4exb/sdrangel #953: Web API update for "Satellite Radio Control" tabs in "Satellite Tracker" feature

Valstee, 2021-07-06:
Hi @f4exb, @srcejon, 
Is it possible to PATCH "Satellite Radio Control" settings (in the Satellite tracker feature) using the web API ?
It will be perfect to control my small "raspberry ground station" with sdrangel server !
Thanks a lot for your answer,
Valentin

srcejon, 2021-07-06:
Not at the moment.  The radio control settings aren't yet accessible.

Valstee, 2021-07-06:
Ok ! Thanks for your answer @srcejon, it could be great  ;-)

srcejon, 2021-07-09:
This should now work. You can add settings with something like:

    curl -X PATCH "http://127.0.0.1:8091/sdrangel/featureset/0/feature/0/settings" -d '{"featureType": "SatelliteTracker",  "SatelliteTrackerSettings": {  "deviceSettings": [ {  "satellite": "ISS",  "deviceSettings": [ { "deviceSet": "R0", "doppler": [0], "frequency": 0, "presetDescription": Sat", "presetFrequency": 145.825, "presetGroup": "ISS Digi", "startOnAOS": 1, "startStopFileSinks": 1, "stopOnLOS": 1}] } ] }}'

Valstee, 2021-07-09:
Yes really good! I confirm you that it works! A huge thanks! 🥇 
Can I close the issue or are there still things to do? (because you said "start to add")

srcejon, 2021-07-09:
Yep, you can close - another patch was needed - but I forgot to tag it with this issue number.

closed 2021-07-12 by Valstee