f4exb/sdrangel #955: Scope stops working after adjusting trace length / time range / offset

srcejon, 2021-07-08:
Hi,

It appears the scope stops working after adjusting the length / time range and offset. To recreate:

- Open channel analyzer (I'm running at sample rate of 128kS/s for this test)
- Increase trace length to  something like 5 seconds
- Reduce time range from 5 seconds to ~100 ms
- Adjust time offset from 0 to ~100ms
- Set time offset back to 0
- Set time range back to 5 seconds

At this point, the scope stops updating and the trace length slider can't be moved.

srcejon, 2021-07-08:
Also, when it gets in this state, if you exit SDRangel, then restart, it can crash on startup as it restores the default preset.

f4exb, 2021-07-08:
Moving the offset automatically switches to memory mode hence the trace length cannot be changed. To recover full control turn the memory button all the way to zero.

srcejon, 2021-07-08:
Sorry, didn't see that comment on the other issue. 

Looks like there is a problem if I try to switch back to trace 0.

Start acquisition
Stop acquisition
Adjust offset
(Trace switches to memory 1)
Switch back to trace 0 (live)

Debugger stops with an exception "cannot seek value-initialized vector iterator"

[image]

f4exb, 2021-07-09:
This is illogical because here: https://github.com/f4exb/sdrangel/blob/master/sdrbase/dsp/scopevis.cpp#L1080 `m_currentTraceMemoryIndex` is 0 and thus `processMemoryTrace()` cannot be called.

I cannot reproduce the issue in Linux (with gcc).

srcejon, 2021-07-09:
I see m_currentTraceMemoryIndex being 1, even though it is 0 in the GUI.

And this is printed in the console:

2021-07-09 16:45:46.646 (D) ScopeVis::handleMessage: MsgScopeVisNGMemoryTrace:  1

It seems the sequence above isn't 100% reproducible for me today. Try this:

Test source at 64k
Add channel analyzer
Start acquisition
Stop acquisition
Adjust trace length to > 10s
Adjust offset
(Trace switches to memory 1)
Switch back to trace 0 (live)
Crash

2021-07-09 16:57:43.334 (D) ScopeVis::handleMessage: MsgScopeVisNGMemoryTrace:  49

is printed in the console.

f4exb, 2021-07-11:
a few strange things:

1. When no decimation takes place in the Channel Analyzer I cannot set the trace length > 10s. The slider goes to a maximum of 100. This is multiplied by the "chunk size" of 4800 to yield 480000 samples. At 64 kS/s this is 480000 / 64000 = 7.5s. This is the maximum I observe.

2. At the last step it behaves very differently:
```
2021-07-11 09:57:45.197 (D) ScopeVis::handleMessage: MsgScopeVisNGMemoryTrace:  0
2021-07-11 09:57:45.197 (D) ScopeVis::setSampleRate: 64000 S/s
2021-07-11 09:57:45.258 (D) GLScopeGUI::onScopeTraceSizeChanged: 480000
2021-07-11 09:57:45.258 (D) GLScopeGUI::onScopePreTriggerChanged: 0
```
It seems that in your case although it should be before in the memory nb 1 position it does as if it was 0. Then when going backwards it wraps around and falls on 49. 

Which version of Qt do you have? I am on 5.15.2,

f4exb, 2021-07-11:
Also when you switch trace back to 0 does it display "M 00" for the text (as it should)?

f4exb, 2021-07-11:
I have looked at the code of `scopevis.cpp`... When the trace memory number in the GUI it passes the instruction to the ScopeVis via message that is handled here: https://github.com/f4exb/sdrangel/blob/master/sdrbase/dsp/scopevis.cpp#L1062 What I notice is that this message handling block is not locked by mutex like some others. I don't know how it may happen but there might be a race condition here. This would also explain that the issue is not 100% reproducible for you. On my side it is ~0% reproducible (couldn't make it yet) so it would be rather difficult to validate.

I think it does not hurt to put a mutex locker there anyway (and maybe in other blocks) so I can push change in that sense.

f4exb, 2021-07-11:
As mentioned in the commit message it "may" fix the issue (to be confirmed).

closed 2021-07-11 by f4exb

f4exb, 2021-07-11:
This commit https://github.com/f4exb/sdrangel/commit/ede1ef015100e6ddc6e0c38c2ad24eb76c565ac6 solves the issue when the channel analyzer / scope is left with some time offset. In fact it cannot be made persistent in the settings.

srcejon, 2021-07-13:
> When no decimation takes place in the Channel Analyzer I cannot set the trace length > 10s.

Sorry - I forgot to mention that I had decimation enabled.

On Windows, I'm on Qt 5.15.1 and Linux 5.14.2.

It seems like the mutex may have helped - as I've tried a few times on Linux without a crash. Thanks.

However, it seems that I do still occasionally get an exception on Windows. 

[image]

I can't seem to come up with a reproducible sequence for it though.

commit 785ff0ed (2021-07-11): ScopeVis: added missing mutex locks in message handling. May fix #955