f4exb/sdrangel #961: Display problem with the pager demodulator

choupette-change, 2021-07-15:
[image]
If I use only one demodulator, I have access to all the settings of the "waveforms" part.

[image]
If I use several (at least 2 simultaneously), I no longer have access to all the possible settings of the "waveforms" window of the demodulator ?? Same problem with Windows and ubuntu 21 04 versions.
Can you help me ?

choupette-change, 2021-07-15:
Same problem with AIS Demodulator. An another question : Is it possible to plan to configure (different color) the scrolling bars, they are difficult to distinguish in strong light or penumbra?

srcejon, 2021-07-15:
The above patch should fix it for Pager, AIS and Radio Clock.

If you want different themes, I'd suggest opening a new issue.

PR #962 (2021-07-15): Increase widget minimum width when scope is visible
So that all of its controls are visible. Fixes #961

(Also a few little typo fixes / cleanups in some other files)
closed 2021-07-15 by f4exb