f4exb/sdrangel #964: Not start metis scan

hiperiondev, 2021-07-15:
Sdrangel can't find any metis device.
If start with QT_LOGGING_RULES="*.debug=true" I don't see any message "DeviceMetisScan::scan: discovery writeDatagram sent" (devicemetisscan.cpp line 50) nor any udp related package in wireshark

f4exb, 2021-07-15:
Not here:
```
2021-07-15 18:36:21.435 (D) DeviceMetisScan::scan: discovery writeDatagram sent
```

f4exb, 2021-07-15:
You must verify if the plugin is correctly loaded and registered. You should also see this in the log:
```
2021-07-15 18:36:18.861 (D) PluginManager::loadPluginsDir: fileName: libmimometismiso.so                                                                                                                                                    
2021-07-15 18:36:18.861 (I) PluginManager::loadPluginsDir: loaded plugin libmimometismiso.so   
```
and
```
2021-07-15 18:36:18.949 (D) PluginManager::registerSampleMIMO  Metis MISO  with MIMO name  sdrangel.samplemimo.metismiso  and hardware id  "MetisMISO"   
```

hiperiondev, 2021-07-15:
My mistake. Firewall was the problem.
Txs

closed 2021-07-15 by hiperiondev