f4exb/sdrangel #967: Wrong address detection and garbage in hebrew

zukik, 2021-07-20:


zukik, 2021-07-20:
[.2021-07-20T11_37_24_997.zip](https://github.com/f4exb/sdrangel/files/6847967/default.2021-07-20T11_37_24_997.zip)

zukik, 2021-07-20:
the pocsag decooder  does not show correct address compair to other two  pocsag  decoders that  the other two  show other sdress which is not same as the  sdrangel show

srcejon, 2021-07-20:
Could you please confirm the centre frequency of the signal I should be looking at? Also, what is the baud rate?

zukik, 2021-07-24:
Sorry for  the delay  i didnt see the  mail  before
freq is 152.240 1200 baud
Thank U  ronen- 4Z4ZQ
http://www.ronen.org

srcejon, 2021-07-25:
When I try this in SDRangel - I see lots of parity errors indicated (which means some received bits are incorrect) in the even PE and BCH PE columns - which would explain why the displayed data isn't correct.

If I pipe audio from the NFM Demod in to PDW - then I also cannot see a consistent decode - each time I replay the file, PDW decodes something else. (Some of the addresses will match the SDRangel decode)

This suggests it's not a clean signal - but also possible something isn't quite right with my setup. Do you get a consistent decode in PDW? If so, do you have a screenshot? 

Are other pager messages received sucessfully?

zukik, 2021-07-26:
I feed  the PDW with  HDSDR and the  sdrangel  is running stand alone
is there any parameters i should notice  in the sdrangel in order to  metch the decode of the sdrangel  ?  i set the  bandwidth of the   sdrangel pocsag decoder to max  and also the deviation setting   but mabe i should do something in the NFM  decoder  ? or it does not   connected and not effected   at the   pocsag decoder ?
As i said  im willing to give acess to this computer with   teamview or  AnyDesk (prefered)  so you can play from remmote and check it
Regards
The signal by the way is very strong  on the spectrum ....
Ronen - 4Z4ZQ

f4exb, 2021-07-27:
In your .2021-07-20T11_37_24_997.zip recording the sampling rate 10Ms/s is way too high for a signal that needs a bandwidth in the 10's kHz. Try to reduce it to focus on your signal of interest. You should not need more than 100 kS/s sampling rate.

This probably results in a lot of samples drop out.

zukik, 2021-07-28:
I  need the 10MHZ full span for the spectrum to hunt signals and this  works fine for the DMR  as an example the DMR  decode has no problem   but ill try it  and let you  know
the HDSDR  that  i pipe its audio  to    the  Virtual audio cable  and the pdw that listen on the virtual audio cable  works with 10MHZ  sample raye and decode excelent  but il check and let you know
Ronen - 4Z4ZQ

zukik, 2021-07-30:
Hi
I   reduced the sample rate and indeed the decode now is same as the PDW decode and also the addresses are the same at  both  software
I have also  corrected the  hebrew table that was not 100 percent correct  and i get now hebrew fine
Thank you for all the help
Keep up the good job
73's
Ronen-4Z4ZQ
http://www.ronen.org<http://www.ronen.org/>

srcejon, 2021-07-31:
What corrections did you make? Adding the extra 5 characters? If so, i have already made that change. If not, please let me know what needs to change.

zukik, 2021-07-31:
Yes  i aded  few characters that was   missing
It didnt gave me to save it but as long as the program is open it work OK
Ronen-4Z4ZQ

closed 2021-09-03 by srcejon