f4exb/sdrangel #972: DVB-S2 over UDP is unusable.

DJRaov, 2021-07-27:
Trying to transmit DVB-S2 with .ts frames from a UDP source results in a unusable stream, with only VLC being able to pick the very chopped up frames. File playback works fine, but is not what I need.

srcejon, 2021-07-27:
Please tell us some more information about the transport stream (e.g. bitrate) etc, what settings you are using for ffmpeg and the modulator and so on. A screenshot might be helpful too.

DJRaov, 2021-07-27:
Any bitrate over 160kb/s is unusable. Content of the transport stream does not matter, be it a single stream or multiple muxed together. Changing modulator settings doesn't do anything either. I think it's a UDP implementation issue, as when writing to a file directly from ffmpeg and trying to stream that file works flawlessly.

srcejon, 2021-07-27:
What settings are you using for ffmpeg and the modulator?

Are you running ffmpeg on the same computer? Windows or linux.

The more information youbgive, the more likely it is I'll be able to help.

DJRaov, 2021-07-28:
For ffmpeg I'm ingesting 4 UDP streams (2 remote and 2 local), transcoding to x264 MP@4.1L, and encapsulating into mpegts. Modulator is running at 3500ksps, S2 mode, QPSK modulation and 3/5 FEC. On the hardware side of things is a HackRF with firmware compiled about a week ago, running at about 7.1MSPS.

srcejon, 2021-07-28:
Ok, but what are the exact command line options you are passing to ffmpeg?

Is this Windows or Linux?

Can you post a screenshot of the modulator so I can see what rate it is receiving packets at.

DJRaov, 2021-07-28:
`ffmpeg -i udp://10.0.0.2:30001 -i udp://10.0.0.2:30002 -i udp://127.0.0.1:30003 -i udp://127.0.0.1:30004 -i udp://127.0.0.1:30005 -map 0:0 -map 1:0 -map 1:1 -map 2:0 -map 2:1 -map 3:0 -map 3:1 -map 4:0 -program program_num=1:title="Current Time":st=0 -program program_num=2:title="omitted1":st=1:st=2 -program program_num=3:title="omitted2":st=3:st=4 -program program_num=4:title="omitted3":st=5:st=6 -program program_num=5:title="EngData":st=7 -metadata service_provider="omitted4" -metadata service_name="TextMux" -f mpegts "udp://127.0.0.1:10000?fifo_size=5000000"
` is the command, I am on Linux. Though, same issue occurs on Windows too. 
[image]

srcejon, 2021-07-28:
Thanks. The UDP bitrate given there is 3509kbit/s. How does that compare to your expected bit rate?

It might be easier to debug, if you stream the .ts file you have there via ffmpeg - then there will be fewer unknowns than combing multiple streams (which I can't easily replicate).

If you can compile from source, you might want to try increasing the UDP buffer size. In datvmodsource.cpp, increase:

            m_udpSocket->setSocketOption(QAbstractSocket::ReceiveBufferSizeSocketOption, 1000000);

the 1000000 to something larger, as at high data rates, packets could be being dropped.

You might also want to experiment with the UDP url - see if pkt_size and bitrate have any effect: udp://127.0.0.1:10000?pkt_size=1316&bitrate=600000"

DJRaov, 2021-07-28:
Increasing the UDP bitrate in the URL helped somewhat, but still not ideal. Looking at the STB, there are random and sudden quality drops all the way down to 0, so some buffer fiddling might also be required. This happens, in my case, on samplerates over 7MSPS. I will also try increasing the UDP buffer.

DJRaov, 2021-07-28:
Update: increased to 3000000, still better, but not perfect. Would it be enough of a pain to implement TCP to not do it?

f4exb, 2021-07-28:
What do you expect from TCP vs UDP? TCP is connection based and re-sends packets if they are not correctly received at the other end so the overhead can only be worse.

DJRaov, 2021-07-28:
More reliability. UDP overall has been a huge pain with what I am working with.

srcejon, 2021-07-28:
Try setting the the buffer to 50M - and then with the improvements from ffmpeg options above (also try -flush_packets 0) what bitrate can you stream via UDP using a ts file as the source successfully at? Could you please send a screenshot of that - then increase the bitrate a bit and send a screenshot of that as well, along with details of what the bitrates are for the TS files.

f4exb, 2021-07-28:
If your network is unreliable or even only reasonably reliable then TCP will be a pain even worse and if it is reliable enough then you don't need the overhead of TCP. UDP is recommended for streaming and you have to live with the fact that when streaming if a packet is lost then it is lost.

DJRaov, 2021-07-28:
The network is definitely reliable, as it is a direct pc-to-pc link over gigabit ethernet. 
If streaming a file over UDP then I can't transmit ANY file and get a video output. TCP overhead is not an issue here, as everything is done over a isolated link.

srcejon, 2021-07-28:
I've added a display of the UDP receive buffer utilization in PR #976. This currently only works on Windows. Please give it a try and then send the screenshots I requested above - so we can see whether receive buffer size is part of the problem.

srcejon, 2021-07-28:
Also, in your screenshot, it shows you've been playing a file and UDP. As a precaution, I'd suggest not playing a file, if using UDP, just in case the two are interfering somehow (they shouldn't - but just something to check).

DJRaov, 2021-07-28:
The file was paused. And yes, they weren't.

DJRaov, 2021-07-29:
So here is what I found after some experimentation: if there is ever a clog somewhere in the processing line and packets start gradually filling the buffer up, once the buffer reaches 100% they are not flushed, bringing the entire transport stream down. This can be mitigated by increasing the uplink budget (higher SRs, higher FEC, switching to 8PSK and even 16APSK), but also brings unnecessary idling. Is it possible to add a "flush buffer if full" toggle?

srcejon, 2021-07-29:
Not really, as is, as it wouldn't work on Linux.

The question is why is the buffer getting full? Does it need to be bigger to support a temporary bump in a variable rate? Or is the input rate higher than the output rate?

If it never recovers, it sounds like the latter.

DJRaov, 2021-07-29:
Input rate is lower than the max output rate available in the signal, and I'd say increase it to 10 or 15 million to mitigate the temporary bump. It recovered after around 20 minutes by the way.

srcejon, 2021-07-29:
As I said above, try setting it to something like 50M or even bigger (so it can't overflow), then run it for a while and see what the max utilisation is, then let me know the result along with a screenshot showing the symbol/data/UDP bitrates. Thanks.

It does sound like you just need to reduce the TS bitrate a little bit though. Also, perhaps try to use CBR rather than VBR (or constrained VBR). There's also an option in ffmpeg that sets the size of the buffer used to calculate the bitrate, so you might want to experiment with that too.

Try something like https://www.videohelp.com/software/Bitrate-Viewer-2 to see how bursty your source is.

DJRaov, 2021-07-29:
I managed to get the TS stream to be very stable, but there are still (small, manageable by the buffer, but nonetheless) spikes in the bitrate. There are still some issues though - the video drops out for about a second every minute or so until the demodulator relocks. (using a hardware STB with a direct connection) I suspect this to be a issue with IQ handling, especially buffers, as I had that happen even when streaming from a file. Could you guys look into that?

srcejon, 2021-07-29:
Do you have a testcase to try? i.e. a TS file and the settings to use.

DJRaov, 2021-07-30:
I can get a .ts file recorded no problem (i'll post it later), modulator settings are as follows: 
DVB-S2 mode
QPSK @ 3000ksps with 2/3 FEC (2000ksps also works with 8/9 FEC)
7MSPS BB rate
UDP input

DJRaov, 2021-07-31:
Update: SDRAngel seems to begin repeating certain BB "chunks" when putting any traffic through to the modulator, confusing the demodulator. Happens on all samplerates, on both Windows and Linux. (buffer never went over 55%)
4MSPS BB and 2000ksps @ QPSK 4/5 work fine when idling, but for the first 3-5 minutes of traffic the lag is pretty intense. Definitely going to blame this on software optimization. ffmpeg settings are as follows: `ffmpeg -i udp://10.0.0.2:30002 -i udp://10.0.0.2:30003 -map 0:0 -map 0:1 -map 1:0 -map 1:1 -program program_num=2:title="omitted1":st=0:st=1 -program program_num=3:title="omitted2":st=2:st=3 -f mpegts -c:v libx264 -profile:v main -level 4.1 -preset superfast -tune animation -crf 24 -b:v 2600k -c:a aac -probesize 3500000 -flush_packets 0 "udp://10.0.0.2:10000?pkt_size=1316&bitrate=4500000"`, input is 2 H264 streams @ 8000kbps, with similar settings to ffmpeg output but with crf of 21.
Output Q values seem to not be fixed despite -crf being set, but that is more of a ffmpeg issue.

srcejon, 2021-07-31:
Can you upload a .ts file that shows the problem?

DJRaov, 2021-07-31:
Now that I have actually learned how to use ffmpeg properly while resolving the issue, I have optimized the transport stream enough for there to be no lag.
Only tried with QO100-allowed rates like 1MSPS and such, but it works fine now.
If anything will pop up I will reopen the issue. Thanks for helping me out!

closed 2021-07-31 by DJRaov