f4exb/sdrangel #977: Problem with HPSDR ethernet interface

hiperiondev, 2021-07-29:
I'm working in HPSDR interface for RPITX (https://github.com/hiperiondev/hpsdr-rpitx)
In particular I use a RPI W Zero with SPI ethernet card.
Sdrangel only recognize connection over wifi but not over ethernet.
Linhpsdr and Quisk works ok.

f4exb, 2021-07-29:
> Sdrangel only recognize connection over wifi but not over ethernet.

Can you develop a bit more? Stated like that this does not make sense. WiFi or Ethernet do not make any difference on the network.

hiperiondev, 2021-07-29:
I agree with you:  this does not make sense.
My test is this: start process in RPI waiting connection (connected over wifi and ethernet on differents networks)
From PC in the same situation (wifi and ethernet) I use  Linhpsdr and Quisk. In this case both programs retrieve the device in two places (net wifi and net ethernet). If I disconnect one I can discover on other connected network.
If I make same test with Sdrangel only in wifi can discover RPI.

hiperiondev, 2021-07-29:
If you have a RPI can make same test with hpsdr simulator: https://github.com/hiperiondev/hpsdrsim

f4exb, 2021-07-29:
It may well be possible that the broadcast packet is sent from just one network interface of the PC (it picks up the WiFi) and thus only devices on this network will reply. Although the broadcast is 255.255.255.255 it will not cross network boundaries.

What happens if you keep just the Ethernet interface up on the PC (i.e. shut down the WiFi)?

hiperiondev, 2021-07-29:
> What happens if you keep just the Ethernet interface up on the PC (i.e. shut down the WiFi)?
Exactly the same thing happens

f4exb, 2021-07-30:
Which means you don't see the RPi at all then?

hiperiondev, 2021-07-30:
In RPI I have a service awaiting for a Metis discovery packet.
Over WIFI this packet arrives ok, but over ethernet never arrives.

With Linhpsdr and Quisk this does not happen

f4exb, 2021-07-30:
OK it is still not clear sorry so let's be even more precise. Can you please shut down the WiFi completely and just use Ethernet. Then does the packet arrives through Ethernet or not?

hiperiondev, 2021-07-30:
> OK it is still not clear sorry so let's be even more precise. Can you please shut down the WiFi completely and just use Ethernet. Then does the packet arrive through Ethernet or not?

That was the first test I did.
If I turn off the WIFI and leave only ethernet, the packets do not arrive

f4exb, 2021-07-30:
So there is nothing to be done about it. I have a Red Pytaya on my Ethernet network that is perfectly recognized so you should check your networks settings instead.

hiperiondev, 2021-07-30:
I'll try. Txs

hiperiondev, 2021-08-07:
I reopen this issue after doing more testing.

Working on another project (https://github.com/hiperiondev/hpsdr-rtx-iq)) I ran into the same problem.

After searching a lot, I found that, when a broadcast is done (and only one) without associating the bind to an interface, it is Linux that chooses which one to exit from, and generally it does so according to the route table.

In particular, choose the interface that has the default route.
In my case it is the wifi, possibly in yours it is ethernet where the RedPitaya hardware is connected.

I tried to change my default route to ethernet and Sdrangel finds the device.

To be able to search in all the interfaces, a broadcat with bind must be sent to each interface, iterating in all the available ones.

commit a1120bca (2021-08-08): Metis scan: scan all network interfaces. Fixes #977
closed 2021-08-08 by f4exb