f4exb/sdrangel #978: Adding a SoapySDR source (UHD B201 + HackRF) on a different subnet

twi42, 2021-07-30:
Hi,

I have a working SoapySDRServer with Ettus B210 and HackRF installed and I can use those flawlessly with SDRAngel over the network when they are automatically probed in the same subnet.

If I'm on a different broadcast subnet, how can I find/probe these SoapySDR devices so that they are recognized in SDRAngel? I tried adding manual SoapySDR entries with "driver=uhd, address=x.x.x.x" but to no avail.

Running SoapySDRUtil in the remote computer (in the different subnet) discovers those sources ok:

SoapySDRUtil --find="remote=tcp://10.1.1.1:55132"

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] [UHD] Win32; Microsoft Visual C++ version 14.2; Boost_107500; UHD_4.0.0.HEAD-0-g90ce6062
Found device 3
  device = HackRF One
  driver = remote
  label = HackRF One #0 909864c82c1f58cf
  part_id = a000cb3c005e4f58
  remote = tcp://10.1.1.1:55132
  remote:driver = hackrf
  serial = 0000000000000000909864c82c1f58cf
  version = 2018.01.1

Found device 4
  driver = remote
  label = B210 310BR0551
  name = 3000501960581
  product = B210
  remote = tcp://10.1.1.1:55132
  remote:driver = uhd
  remote:type = b200
  serial = 310BR0551

Great work with SDRAngel!

Best regards,
Pasi

f4exb, 2021-07-31:
I think this is just "address=x.x.x.x". At least this is how it works with redpitaya. In fact it knows the driver already.

twi42, 2021-07-31:
Hi,

When copying SoapySDR instance with "address=x.x.x.x" I only get an empty device, so HackRF and B210 are not discovered. On the SoapySDR side I also don't see an incoming connection when starting SDRAngel or the device.

closed 2021-12-18 by github-actions[bot]