f4exb/sdrangel #988: Add feature on sdrangel server

Valstee, 2021-08-19:
Hi @f4exb ,
I can't add a feature when I use sdrangel server.
What i do :
- Run sdrangel server with 16bit Rx samples in a docker container.

and then :
- Open Rx0 with POST  --> /deviceset?direction=0
- Set RTLSDR on Rx0 with PUT --> /deviceset/0/device
- Setup my RTLSDR on Rx0 with PUT /deviceset/0/device/settings

and finally :
- Try to add a feature with POST --> /featureset/0/feature

but i receive an error 404 :   "message": "There is no feature set with index 0"
It works with the GUI version but not with the server version.
Do you know why ?

Thanks,

Valentin

f4exb, 2021-08-19:
You need to add a JSON payload with the feature type like:
```json
{
  "featureType": "AIS"
}
```
Probably code 400 would be more relevant than 404 maybe it is not taking the right error path.

Valstee, 2021-08-19:
Yes, my JSON payload is :
`
[
        "/featureset/0/feature",
        "POST",
        null,
        {
            "featureType": "GS232Controller",
            "direction": 0
        },
        "Add GS232Controller"
    ]`

f4exb, 2021-08-19:
Then put exactly this for the payload
```json
{
  "featureType": "GS232Controller"
}
``` 
This will create a GS232Controller feature with default settings. Then in a second call you will adjust the settings

Valstee, 2021-08-19:
Many thanks for this detailed answer. This is exactly what I am doing !

However even with the interractive documentation swagger UI this curl :

`curl -X GET "http://192.168.80.21:8091/sdrangel/featureset/0" -H "accept: application/json"`

gives me this answer error 404 :
`{
  "message": "There is no feature set with index 0"
}`

I can't understand why with the server version I have this error when everything works with the GUI version

f4exb, 2021-08-19:
OK I was trying with a GUI version. What happens is that GUI and Server do not use the same code for the early processing of the request. This is indeed a bug in the Server version.

f4exb, 2021-08-19:
OK this is with the Server version. I think it is missing the addition of a FeatureSet completely. This is similar to DeviceSets. The GUI always starts with a DeviceSet and also a FeatureSet but this is not the case in the server. So in fact it is missing the equivalent of POST `/sdrangel/deviceset` thus POST `/sdrangel/featureset`.

Valstee, 2021-08-20:
Yes I think you are targeting the right problem !
I tried to do something similar to creating a deviceset but for a featureset before making this ticket. It didn't work... 
So when you say that a POST `/sdrangel/featureset` is missing. It's in what I'm doing ? or it's not yet implemented in sdrangel ?

f4exb, 2021-08-20:
It is not implemented and needs be

f4exb, 2021-08-21:
GET `/sdrangel/featuresets` does not work either and will be fixed with this issue

f4exb, 2021-08-21:
Now on top of master at commit 600fbeb48656cd54b4373e49de2e0aee8838cefd

f4exb, 2021-08-22:
Implemented in v6.16.4

closed 2021-08-22 by f4exb

Valstee, 2021-08-23:
Thank you very much for your reactivity! 
This is exactly what I needed for my raspberry receiver station! Awesome !!

f4exb, 2021-08-23:
I knew this was for a good cause 😉