f4exb/sdrangel #996: API - Error 404 "There is no feature with id SatelliteTracker" with sdrangel SERVER

Valstee, 2021-09-01:
Hi @f4exb, 
With SDRangelSRV and this configuration :
```
"appname": "SDRangelSrv",
"architecture": "arm",
"os": "Debian GNU/Linux 10 (buster)",
"qtVersion": "5.11.3",
"version": "6.16.4-3-g8d68fa523"
```
I've got an error 404 - "There is no feature with id SatelliteTracker" when i try to Add **Satellite Tracker** feature with this request :
`http://127.0.0.1:8091/sdrangel/featureset/0/feature`
`{"featureType": "SatelliteTracker"}`
Do you have the same issue ?
Thanks,
Valentin

f4exb, 2021-09-02:
My guess is that for some reason the Satellite Tracker plugin is not loaded and registered. Normally in the log when starting SDRangel you should see these lines:
```
2021-09-02 05:07:52.881 (D) PluginManager::loadPluginsDir: fileName: libfeaturesatellitetracker.so                                                                                                                                          
2021-09-02 05:07:52.894 (I) PluginManager::loadPluginsDir: loaded plugin libfeaturesatellitetracker.so   
```
And ...
```
2021-09-02 05:07:54.560 (D) PluginManager::registerFeature  Satellite Tracker  with channel name  "sdrangel.feature.satellitetracker" 
```
BTW there was a copy/paste error in the log message above it should say: "with feature name..."

f4exb, 2021-09-02:
Note: see this: https://github.com/f4exb/sdrangel/blob/master/plugins/feature/CMakeLists.txt#L22

Valstee, 2021-09-02:
Thanks a lot for your detailed answer !
So I add `-DSGP4_DIR=/opt/install/sgp4 \` here : https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/Dockerfile#L520 to create my docker image.
But this error appears : 
```
#63 17.65 CMake Error at plugins/feature/satellitetracker/CMakeLists.txt:61 (add_library):
#63 17.65   Target "featuresatellitetrackersrv" links to target "Qt5::Positioning" but
#63 17.65   the target was not found.  Perhaps a find_package() call is missing for an
#63 17.65   IMPORTED target, or an ALIAS target is missing?
#63 17.65
#63 17.65
#63 17.65 CMake Error at plugins/feature/satellitetracker/CMakeLists.txt:61 (add_library):
#63 17.65   Target "featuresatellitetrackersrv" links to target "Qt5::Charts" but the
#63 17.65   target was not found.  Perhaps a find_package() call is missing for an
#63 17.65   IMPORTED target, or an ALIAS target is missing?
```
Even if libqt5charts5-dev (5.15.2-2) and qtpositioning5-dev(5.12.2+dfsg-2) are already installed.
I tried to understand what's going on here :https://github.com/f4exb/sdrangel/blob/master/plugins/feature/satellitetracker/CMakeLists.txt#L55
 but I couldn't figure out the error when comparing with the architecture of the other features

f4exb, 2021-09-02:
besides this I see that my official Dockerfile for ARMv8 is not building this feature either: https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/armv8.ubuntu.Dockerfile.

Also there is a few dependencies checks missing in https://github.com/f4exb/sdrangel/blob/master/plugins/feature/CMakeLists.txt#L22 :
  - Qt5Positioning_FOUND
  - Qt5Charts_FOUND
  - Qt5TextToSpeech_FOUND (for GUI)

f4exb, 2021-09-02:
Well there is a glitch somewhere because when I try to re-run the compilation of features in the Docker image for ARMv8 I get:
```
-- Checking for module 'sgp4'
--   No package 'sgp4' found
-- SGP4 LIBRARIES SGP4_LIBRARIES-NOTFOUND
-- SGP4 INCLUDE DIRS SGP4_INCLUDE_DIR-NOTFOUND
-- Could NOT find SGP4 (missing: SGP4_LIBRARIES SGP4_INCLUDE_DIR)
```
Although sgp4 has been built and the location specified in the cmake line.

Valstee, 2021-09-02:
So I don't really understand why but if i start sdrangel server in a docker container with this :
https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/start_server.sh
Satellite Tracker plugin is loaded and registered without adding `-DSGP4_DIR=/opt/install/sgp4 \` here : https://github.com/f4exb/sdrangel-docker/blob/master/sdrangel/Dockerfile#L520
Can this help you to understand what append ?

f4exb, 2021-09-02:
You do have to add `-DSGP4_DIR=/opt/install/sgp4` and then you fall in the next trap. 

For this next trap I think I have found the problem here: https://github.com/f4exb/sdrangel/blob/master/CMakeLists.txt#L329
It is not looking for `Qt5::Positioning` and `Qt5::Charts` that are also required to compile the Satellite Tracker plugin in server version

f4exb, 2021-09-02:
It seems the latest commit fcf22da4a698ddf8f96fe13793a5e40dc833f6eb fixes the issue.

f4exb, 2021-09-06:
Implemented in v6.16.5

closed 2021-09-06 by f4exb

Valstee, 2021-09-06:
Sorry for my late reply.
I had not seen the second trap indeed with the changes you mention, everything works perfectly!
The last thing I need to fix is the automatic saving of File Sink files but I get this error ... 
`2021-09-03 18:41:53.661 (W) FileRecord::startRecording: failed to open file:  "/home/pi/Documents/radio/noaa_record/record.2021-09-03T16_41_53_661.sdriq"`
anyway, this is another topic and I think it's not smart to save the .sdriq files on a raspberry but I'd like to have fun testing different "Noaa demodulator" ... thanks for your help, I should soon have an automatic micro station for the noaa images (and other satellites) and maybe even autonomous with battery/solar panel! ;-)