f4exb/sdrangel #999: Crashes / where is the config?

sgofferj, 2021-09-08:
I'm having a problem with the latest release under Ubuntu 20.04. I have a channel analyzer channel open and when I try to close it, sdrangel crashes. I would like to wipe the config but I couldn't find where sdrangel stores its config data.
Please advise.

srcejon, 2021-09-08:
$HOME/.config/f4exb/SDRangel.conf I think

sgofferj, 2021-09-08:
Indeed, there it is... Thanks!

sgofferj, 2021-09-10:
Wiping the config solved the immediate issue but I tried again and at least on my system, when I add a channel analyzer, I can't get rid of it any more. When I close it, SDRA crashes and after re-opening, the channel analyzer is back. Using latest release on Ubuntu 20.04 LTS.

srcejon, 2021-09-10:
It works for me.

You could try:

$ sudo apt-get install gdb
$ gdb /opt/install/sdrangel/bin/sdrangel
(gdb)run

Then after it crashes:
(gdb) info stack

And post the output here.

closed 2021-12-18 by github-actions[bot]