f4exb/sdrangel #1000: Deserialization error with Sdrangel server

Valstee, 2021-09-09:
Hi @f4exb, 
My setup is the following : sdrangel SRV (server16) on a Raspberry Pi4. 
I have a systematic error when I load a new preset. In my example with atpdemod : 
```
2021-09-09 13:19:26.182 (D) DeviceSet::loadChannelSettings: Loading preset [satellites | noaa_preset]
2021-09-09 13:19:26.183 (D) DeviceSet::loadChannelSettings: 2 channel(s) in preset
2021-09-09 13:19:26.183 (D) DeviceSet::loadChannelSettings: channels compare [sdrangel.channel.filesink] vs [sdrangel.channel.aptdemod]
2021-09-09 13:19:26.184 (D) DeviceSet::loadChannelSettings: deserializing channel [sdrangel.channel.aptdemod]
sdr@raspberrypi:~$
```
I use the sattracker feature to receive noaa and meteor images. When I am on Meteor M2, I use only file sink and for noaa : file sink + aptdemod but it crashes systematically when the noaa preset (file sink + aptdemod) is loaded.
I didn't find how to solve this problem...

srcejon, 2021-09-10:
How are you creating the preset? Would be useful if you could paste the commands so I can recreate it.

Valstee, 2021-09-10:
First I use Sdrangel server (last version) in a docker container by running `./run.sh -f server16` and I use this config file : https://github.com/f4exb/sdrangel/blob/master/scriptsapi/config.py with some .json :
I initialize sdrangel server :
```
[
      [
        "/deviceset?direction=0",
        "POST",
        null,
        "",
        "Open RX 0"
    ],
    [
        "/deviceset/0/device",
        "PUT",
        null,
        {
            "hwType": "RTLSDR"
        },
        "Set RTLSDR on Rx 0"
    ],

    [
        "/deviceset/0/device/settings",
        "PUT",
        null,
        {

            "deviceHwType": "RTLSDR",
            "direction": 0,
            "rtlSdrSettings": {
              "agc": 0,
              "biasTee": 0,
              "centerFrequency": 446012000,
              "dcBlock": 1,
              "devSampleRate": 2200000,
              "fcPos": 2,
              "gain": 197,
              "iqImbalance": 0,
              "iqOrder": 1,
              "loPpmCorrection": 0,
              "log2Decim": 5,
              "lowSampleRate": 0,
              "noModMode": 0,
              "offsetTuning": 0,
              "reverseAPIAddress": "127.0.0.1",
              "reverseAPIDeviceIndex": 0,
              "reverseAPIPort": 8888,
              "rfBandwidth": 350000,
              "transverterDeltaFrequency": 0,
              "transverterMode": 0,
              "useReverseAPI": 0
            }
          },
        "Setup RTLSDR on Rx 0"
],
```
Then to create NOAA satellites preset, i use this .json : 
```
[
    [
        "/deviceset/0/channel",
        "POST",
        null,
        {
            "channelType": "FileSink",
            "direction": 0
        },
        "Add FileSink"
    ],
    [
        "/deviceset/0/channel/0/settings",
        "PATCH",
        null,
        {
        "FileSinkSettings": {
        "fileRecordName": "/home/sdr/record.sdriq",
        "spectrumSquelch": -100
        },
        "channelType": "FileSink",
        "direction": 0
        },
        "Setup File Sink on channel 0"
    ],
    [
        "/deviceset/0/channel",
        "POST",
        null,
        {
            "channelType": "APTDemod",
            "direction": 0
        },
        "Add APT Demodulator"
    ],
    [
        "/deviceset/0/channel/1/settings",
        "PATCH",
        null,
        {
        "APTDemodSettings":{
        "autoSave": 1,
        "autoSavePath" : "/home/sdr"
        },
            "channelType": "APTDemod",
            "direction": 0
        },
        "Setup APT Demodulator"
    ],
    [
        "/preset",
        "POST",
        null,
        {
  "deviceSetIndex": 0,
  "preset": {
    "groupName": "satellites",
    "type": "R",
    "name": "noaa_preset"
  }
},
        "Noaa preset save"
    ]
]
```
and for METEOR M-2 (after closing the two previous channels) :
```
[
    [
        "/deviceset/0/channel",
        "POST",
        null,
        {
            "channelType": "FileSink",
            "direction": 0
        },
        "Add FileSink"
    ],
    [
        "/deviceset/0/channel/0/settings",
        "PATCH",
        null,
        {
        "FileSinkSettings": {
        "fileRecordName": "/home/sdr/record.sdriq",
        "spectrumSquelch": -100
        },
        "channelType": "FileSink",
        "direction": 0
        },
        "Setup File Sink on channel 0"
    ],
    [
        "/preset",
        "POST",
        null,
        {
  "deviceSetIndex": 0,
  "preset": {
    "groupName": "satellites",
    "type": "R",
    "name": "meteor_preset"
  }
},
        "meteor preset save"
    ]
]
```
Then I register these presets in satellite tracker and when I switch from meteor to noaa, sdrangel srv stops.
But you can reproduce this manually by switching from the meteor preset to the noaa preset (using sdrangel client for example).

f4exb, 2021-09-26:
I think the `config.py` script is rather this one: https://github.com/f4exb/sdrangel/blob/master/swagger/sdrangel/examples/config.py it is an earlier version where data is positional. It works but is a bit less practical.

Anyway, I could reproduce with the x86 version this is related to the server version but not the architecture. In the server version the program passes through this line: https://github.com/f4exb/sdrangel/blob/master/sdrbase/device/deviceset.cpp#L151 and I think it misses initialization to nullptr. Then later it is mistaken for a valid pointer: https://github.com/f4exb/sdrangel/blob/master/sdrbase/device/deviceset.cpp#L173

commit 3e6fe8af (2021-09-26): Fixed pointer initialization. Fixes #1000
closed 2021-09-26 by f4exb