f4exb/sdrangel #1002: Simple VOX TX control

hitsfm, 2021-09-13:
It would be great to have a simple VOX transmit mode on the modulators. Where when input "mic" volume goes above X threshold the transit automatically holds, This would allow us for experimenting with repeater like setups using other sdr units and or software combos on the RX / internet links etc configured output using sound card virtual cables etc..

Thanks for this. Great work.

f4exb, 2021-11-18:
At the risk of making the term "simple" in the SimplePTT feature plugin a bit abusive maybe this could be an addition to it as it relates closely to the same functionality. Thus this would be rather an enhancement to the PTT feature.

f4exb, 2022-01-03:
Implemented in v6.17.7

closed 2022-01-03 by f4exb