f4exb/sdrangel #1003: How to connect to remote Soapy server?

synthead, 2021-09-19:
I am having a hard time figuring out how to connect to my remote Soapy server.  I have SDRangel compiled with Soapy support on Arch Linux, linked to the lib provided by the `soapysdr` package.

On the client side, I can query the Soapy server just fine:

```
$ SoapySDRUtil --find="remote=<host>"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

18:06:04.015390 WARN:   xtrxllpciev0_discovery:243 [PCIE] XTRX PCIe driver isn't loaded
Found device 0
  driver = remote
  label = SDRplay Dev0 RSP1A <serial>
  remote = tcp://<host>:55132
  remote:driver = sdrplay
  serial = <serial>
```

I can also connect to this Soapy server with gqrx without issues.  I am using this device string:

```
soapy=0,driver=remote,remote=tcp://<host>:55132
```

However, I can't seem to get SDRangel to connect to this server.  SoapySDR doesn't show up in the hardware list from Preferences > Devices > User arguments:

[image]

I can kind of "force" it to use SoapySDR by typing "SoapySDR" into the "HwID" text field, clicking the "+" on the right, then changing the Arg string, like so:

[image]

From here, after restarting SDRangel, I see this in the sampling device list:

[image]

However, I get this dialog box when clicking the "play" button with this source:

[image]

When I click the play button, this line is displayed from the console that I started SDRangel:

```
2021-09-18 19:14:27.692 (W) FileInput::start: file not open. not starting
```

What am I doing wrong?

synthead, 2021-09-19:
Related to https://github.com/f4exb/sdrangel/issues/583

closed 2021-12-24 by github-actions[bot]

kevinelliott, 2022-01-23:
I can't get this to work either.