f4exb/sdrangel #1004: ValueDialZ::setValueRange sets value to 0

srcejon, 2021-09-20:
It seems calling ValueDialZ::setValueRange clears the existing value to 0, if it is within the valid range.

	if(m_value < m_valueMin)
	{
		setValue(m_valueMin);
	}
	else if(m_value > m_valueMax)
	{
		setValue(m_valueMax);
	}
	else
	{
	    // Why this here?
            m_text = formatText(0);
	    m_textNew = m_text;
	    m_value = 0;
	    m_valueNew = m_value;
	    update();
	}

This seems like a bug, as none of the other Qt widgets change an in range value when setting the range - and 0 may be out side of the valid range.

If you try to save/restore the value around the call to setValueRange as a workaround, it results in the scrolling animation.

f4exb, 2021-09-26:
This is old code and indeed a bit strange. If the value is in range and you don't change it there should not be the need of an animation.

srcejon, 2021-09-30:
Another issue is that emit changed(m_valueMax/Min) isn't called when the value is clipped to be within range.

Perhaps the calls to emit should be moved inside the setValue function?

commit 40a1c854 (2021-10-03): ValueDialZ: removed set value to 0 code if value is in range. Fixes #1004
closed 2021-10-03 by f4exb