f4exb/sdrangel #1028: API only appears to bind to one IP address

srcejon, 2021-11-01:
It seems that the API only binds to a single IP address (which by default, is 127.0.0.1). If your machine has an address of 192.168.0.1, it will by default only respond to 127.0.0.1. If you use --api-address, it will respond to 192.168.0.1, but not both.

I think it would be useful to support binding to any address as well as just a specific address. This would mean the API can work on machines that have multiple IP addresses (E.g. 192.168.0.1, 192.169.0.1 and 127.0.0.1), but also if we have this option to be the default, the API and remote sink would be able to be used remotely without passing any command line options (which is how I stumbled across this).

The change to support this, seems fairly simple: we just need to have the initial value of MainParser::m_serverAddress and m_serverAddressOption to be "" rather than "127.0.0.1", as "" will bind to any address.

If it is defaulting to only 127.0.0.1 for security reasons, then I guess an alternative implementation would be to change the --api-address option to support "" as an allowed value.

f4exb, 2021-11-01:
I haven't tried this but `--api-address 0.0.0.0` will probably do what you want although this would not be the default. Security should not be considered a concern. In most situations it is expected that the API is behind a firewall of some sort.

srcejon, 2021-11-01:
Thanks - just gave 0.0.0.0 a try and it works on Windows. So could default to that or "".

I can change https://github.com/f4exb/sdrangel/pull/1029 to whatever you prefer - one other change needed is passing a useable value to the About dialog, so the link there works.

closed 2021-11-02 by srcejon