f4exb/sdrangel #1031: Remote Input multicast

srcejon, 2021-11-01:
I'm trying to use the RemoteInput sample source (on Windows). It seems that I can only start acquisition if the Mcast button is checked. Is this the expected behaviour? The readme doesn't describe why I would/wouldn't need to join. (If it is always needed, why have a button for it?)

f4exb, 2021-11-01:
If the Mcast button is set then `QUdpSocket` `joinMulticastGroup` is invoked (in `remoteudphandler.cpp`) and with the address set next to this button.

If you don not want to use multicast this should be possible. However I have never tried this in Windows. Of course addresses between the remote instance and the local instance should match. That is on the remote the address and port of the local instance should be specified and this address and port should be specified in the "Data" field of the local instance.

srcejon, 2021-11-02:
It seems I was just missing pressing the return key after entering the data address:

Example sequence:

Data address is set to 127.0.0.1
Change Data address to real IP address (169.168.0.34)
Don't press return, but change focus with tab or mouse
Set button does not go green - no data being received

Press MCast - Set button goes green
Press Set - data starts to be received
Press MCast - Set button goes green
Press Set - data stops being received

This was the point I opened this issue. However, if I:

Click in Data address and press return - Set button goes green
Press Set - data starts to be received

I see now that the readme says in 8.3 that return key needs to be pressed, but perhaps editingFinished() signal could be used instead of returnPressed() - that captures both return being pressed and loss of focus.

E.g. on_dataAddress_returnPressed -> on_dataAddress_editingFinished and likewise for the others.

closed 2021-11-03 by srcejon