f4exb/sdrangel #1039: SDRangel continuously crashing

Davis0308, 2021-11-04:
I'm running SDRangel on Windows 10 with an RTL-SDR V3. Since the first opening of the software I noticed that by interacting with it, it often crashed for no apparent reason. At times it works for hours, and at times it crashes once and then I have to try restarting it various times to make it work again. I noticed that the majority of crashes happen when I:
-Start the acquisition, stop the acquisition and then start it once more
-Start a plugin
-Change frequency from the dial
---
I was able to go in event viewer and find the details of some of the crashes. This is what it says:

Crash 1:

> Faulting application name: sdrangel.exe, version: 0.0.0.0, time stamp: 0x617518d3
> Faulting module name: Qt5Core.dll, version: 5.15.2.0, time stamp: 0x5fa4dd3b
> Exception code: 0xc0000005
> Fault offset: 0x00000000001d69db
> Faulting process ID: 0xe1c
> Faulting application start time: 0x01d7d197c6c9f0e3

Crash 2:

> Faulting application name: sdrangel.exe, version: 0.0.0.0, time stamp: 0x617518d3
> Faulting module name: msvcrt.dll, version: 7.0.19041.546, time stamp: 0x564f9f39
> Exception code: 0xc0000005
> Fault offset: 0x000000000005ce00
> Faulting process ID: 0x3600
> Faulting application start time: 0x01d7d18b3a0f0906

---
Hopefully a solution to the problem will be found, because this is an awesome piece of software, and I'd love to use it without this inconvenience.

Davis0308, 2021-11-09:
> Do you have similar problems with other SDR software?
> 
> Does moving the SDR to a different USB port or external powered hub make any difference?
> 
> What device settings are you using? A screenshot might help.

I have not had these kind of problems with other SDR software, or if there were any they were minor crashes that didn't reoccur;

I have tried multiple ports, both on the front panel of the case and directly in the motherboard's USB ports, they made no difference;

I'm not sure what you mean by device settings (I guess it's the window labeled R0 where I change frequency?), I don't have much experience with the software yet, but I'll try to include as much as possible in the following screenshots:

[image]

[image]
---
While taking some screenshots I discovered the logging feature, and I used it. I managed to get a crash through the playback/stop button (note: that's not the only usual cause of a crash, but it's the easiest one to replicate) and this is the result:

```
2021-11-09 20:39:11.017 (D) SpectrumVis::handleMessage: DSPSignalNotification:  centerFrequency:  0  sampleRate:  600000
2021-11-09 20:39:11.027 (W) QOpenGLTexture::setData() tried to set a null image
2021-11-09 20:39:11.664 (D) RTLSDRInput::handleMessage: MsgStartStop:  stop
2021-11-09 20:39:11.664 (D) DSPDeviceSourceEngine::stopAcquistion
2021-11-09 20:39:11.664 (D) DSPDeviceSourceEngine::handleSynchronousMessages:  DSPAcquisitionStop
2021-11-09 20:39:11.664 (D) DSPDeviceSourceEngine::gotoIdle
2021-11-09 20:39:11.668 (C) RTLSDRThread: async error: Bad file descriptor
```
I don't know if that could be of any help.
If anything else is needed, I will provide it upon request. Thank you for answering!

srcejon, 2021-11-09:
Yep, those were the settings I was referring to - and they look fine. I've just pressed start/stop 50 odd times without a problem though.

The 'async error: bad file descriptor' comes from the following lines of code:

    if((res = rtlsdr_read_async(m_dev, &RTLSDRThread::callbackHelper, this, 32, FCD_BLOCKSIZE)) < 0) {
        qCritical("RTLSDRThread: async error: %s", strerror(errno));

SDRangel requests some data from the RTL SDR driver, but the descriptor (that identifies which device to read from) is no longer valid. This could be a RTL SDR driver problem or problem in the h/w. Do you hear the USB device disconnect sound when this happens? Do you need to reinsert the RTL SDR device to get it to work again, or just restart SDRangel?

Does it make any difference if you reduce the sample rate from 2400000 to 1000000?

Davis0308, 2021-11-09:
I don't hear anything disconnect when it crashes, and I can just re-open SDRangel and press play again and it works fine;

Reducing the sample rate to 1000000 unfortunately makes no difference, it crashes in the same way;

In case it's needed, I'm going to provide the following pic taken from Device Manager:
[image]

and also provide my system configuration:
```
CPU: AMD Ryzen 5 2600x
GPU: Radeon RX590
Mobo: Gigabyte Aorus X470 Ultra Gaming
C Drive: Intel 660p M.2 NVME SSD
16GB DDR4 3000 Corsair Vengeance LPX
```

srcejon, 2021-11-10:
Is there any possibility you aren't using the rtlsdr.dll that comes with SDRangel?

I just tried librtlsdr.dll, renaming it to rtlsdr.dll, and while it does work a bit, it can crash when stopping acquisition.

Davis0308, 2021-11-10:
I don't think that's the case as I've only downloaded the releases on this very repo, I also tried other versions with no success

Davis0308, 2021-11-10:
I just tried using yet another program's rtlsdr.dll file and now it seems to be actually working, with proper audio reproduction and no crashes after repeating the previous crash conditions. Only different thing I noticed is that when I press stop, it stops, continues playing for a few fractions of a second and then actually comes to a complete stop. Once I press play I see this line on the waterfall and hear white noise for a fraction of a second:
[image]
I don't think it's a real problem but I thought it was worth mentioning, because that doesn't happen with the original rtlsdr.dll.

srcejon, 2021-11-10:
Ok - exactly which rtlsdr.dll are you using? Right click Properties > Details in Windows Explorer should tell you.

Davis0308, 2021-11-10:
I'm using this:
[image]

GhosTmodE2120, 2021-12-23:
Davis0308  where did u get that can i get a copy?

Davis0308, 2021-12-24:
[alternative_rtlsdr.zip](https://github.com/f4exb/sdrangel/files/7771910/alternative_rtlsdr.zip)

this is the one I'm using, hopefully it works for you too

GhosTmodE2120, 2021-12-24:
yes thank you is working great

combatlord, 2022-03-13:
Have the exact same issues as David described in the lates Build 6.20.1.
When i use his rtlsdr.dll all works fine.
I also tried the dll from Issue #1105 which winerrace provided.
This also works fine.

closed 2022-05-19 by github-actions[bot]

BlueEther, 2022-07-13:
Another user here that swapped out the rtlsdr.dll for the one posted here and all the stability issues seem to be resolved 
Build 7.4.0