f4exb/sdrangel #1040: Compilation fail at radioastronomy

justindthomas, 2021-11-07:
I'm working on updating SDRangel on my Raspberry Pi 400. I've had this installed for a while, but haven't done much with it recently and am re-building it all using an SSD. I'm stuck at this point:

~~~~~
[ 45%] Building CXX object pluginssrv/channelrx/demodfreedv/CMakeFiles/demodfreedvsrv.dir/freedvplugin.cpp.o
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp: In member function ‘void RadioAstronomy::callOnStartTime(void (RadioAstronomy::*)())’:
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp:363:47: error: ‘class QTimer’ has no member named ‘callOnTimeout’
         m_sweepTimerConnection = m_sweepTimer.callOnTimeout(this, f);
                                               ^~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp: In member function ‘void RadioAstronomy::sweep1()’:
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp:461:47: error: ‘class QTimer’ has no member named ‘callOnTimeout’
         m_sweepTimerConnection = m_sweepTimer.callOnTimeout(this, &RadioAstronomy::waitUntilOnTarget);
                                               ^~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp: In member function ‘void RadioAstronomy::waitUntilOnTarget()’:
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp:507:51: error: ‘class QTimer’ has no member named ‘callOnTimeout’
             m_sweepTimerConnection = m_sweepTimer.callOnTimeout(this, &RadioAstronomy::sweepStartMeasurement);
                                                   ^~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp:514:51: error: ‘class QTimer’ has no member named ‘callOnTimeout’
             m_sweepTimerConnection = m_sweepTimer.callOnTimeout(this, &RadioAstronomy::waitUntilOnTarget);
                                                   ^~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp: In member function ‘void RadioAstronomy::sweepNext()’:
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp:564:55: error: ‘class QTimer’ has no member named ‘callOnTimeout’
                 m_sweepTimerConnection = m_sweepTimer.callOnTimeout(this, &RadioAstronomy::sweep1);
                                                       ^~~~~~~~~~~~~
/opt/build/sdrangel/plugins/channelrx/radioastronomy/radioastronomy.cpp:575:51: error: ‘class QTimer’ has no member named ‘callOnTimeout’
             m_sweepTimerConnection = m_sweepTimer.callOnTimeout(this, &RadioAstronomy::sweep1);
                                                   ^~~~~~~~~~~~~
[ 45%] Linking CXX shared library ../../../lib/pluginssrv/libsigmffilesinksrv.so
[ 45%] Building CXX object pluginssrv/channelrx/radioastronomy/CMakeFiles/radioastronomysrv.dir/radioastronomyplugin.cpp.o
make[2]: *** [pluginssrv/channelrx/radioastronomy/CMakeFiles/radioastronomysrv.dir/build.make:76: pluginssrv/channelrx/radioastronomy/CMakeFiles/radioastronomysrv.dir/radioastronomy.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
~~~~~

I'm following the compilation directions as I've done in the past, and have all the recommended libraries (including the QT5 stuff) installed. It looks to me like it's finding the QTimer library and it _should_ have a member named `callOnTimeout`, so it's unclear to me why the compile is failing.

-Justin

N5BRG, 2021-11-07:
Have a look here and see if it helps you:

https://github.com/thehookup/Holiday_LEDS/issues/5

Bob N5BRG

f4exb, 2021-11-07:
https://doc.qt.io/qt-5/qtimer.html#callOnTimeout

> This function was introduced in Qt 5.12

Qt >= 5.12 is required

justindthomas, 2021-11-07:
> https://doc.qt.io/qt-5/qtimer.html#callOnTimeout
> 
> > This function was introduced in Qt 5.12
> 
> Qt >= 5.12 is required

Okay, thanks. I reverted to commit c623f337d20c467c7aa0b1c8ad200ff8c14b1d2a - just before `radioastronomy` was added and was able to get it all compiled and running. I'll work on updating the Qt libraries on this machine. The basic Raspbian install seems to be stuck on 5.11, but I think I can compile and install newer versions manually.

closed 2021-11-07 by justindthomas

f4exb, 2021-11-07:
I am not following Raspbian closely but I think it is Debian based and you will need the "Bullseye" revision that jumped from Qt 5.11 ("Buster") to Qt 5.15. The Docker image build was based on Buster and I remember upgrading the base image to Bullseye for this reason.