f4exb/sdrangel #1041: no access to load the file ??

choupette-change, 2021-11-07:
[image]

srcejon, 2021-11-07:
In the log, I see:

2021-11-07 12:05:59.169 (C) HttpDownloadManager: SSL error:  "The issuer certificate of a locally looked up certificate could not be found"
2021-11-07 12:05:59.169 (C) HttpDownloadManager: SSL error:  "The certificate has expired"

So a problem on the https://iaru-r1-c5-beacons.org server.

f4exb, 2021-11-07:
Why do you open an issue in SDRangel then?

srcejon, 2021-11-09:
I didn't?

I guess I can improve the error message in the GUI though.

f4exb, 2021-11-09:
I thought the comment was from the reporter. Yes a more informative message could tell the user where the actual issue is.

closed 2022-02-14 by github-actions[bot]