f4exb/sdrangel #1043: AISMod TX API changes

srcejon, 2021-11-12:
Hi,

Regarding 006da4e87290c32531dcde4c1666f53e10368368, the idea of having the encoding in AISMod rather than the GUI is sensible, but can the API be kept consistent with the other modulators (i.e. PacketMod and 802.15.4) so that the TX action that takes a data array, allowing an arbitrary frame to be transmitted with a single API call? Transmitting a stored message could be a different action. Thanks.

f4exb, 2021-11-12:
In fact this is the result of my own experience with the API when implementing it in SDRangelcli https://github.com/f4exb/sdrangelcli (not available yet). In the case of AISMod I definitely needed to have "encode" as an action and therefore handled in the AISMod class. Apparently this is not the case with PacketMod (still under work). As a result the "m_data" of the settings structure is populated and then there is no need to pass it in the "tx" action. Do you have a concrete use case where passing the data as string in the tx action is necessary? I see no issue in first trigger the "encoding" to obtain data and then transmit it. You may also put your own data and transmit it. Therefore I thought passing it in the "tx" action would be redundant or at least confusing.

Packet Modulator is similar although a bit different since there is no "encoding" action. The various parts are passed and assembled into the final message. However similarly this is redundant with what is found in the settings structure.

srcejon, 2021-11-15:
I find it more convenient to send just a single command, but that's a touch subjective.

While it doesn't have a big impact on AIS due to the low data rate, for 802.15.4, having to use two API calls instead of one, will reduce throughput.

f4exb, 2021-11-15:
For high rates the http API performance is prohibitive anyway. One should rather rely on UDP.

f4exb, 2021-11-18:
In the end I will leave the "data" in the 802.15.4 mod actions but turn the actions in a flat structure with "tx" key as a (pseudo) boolean (although tx: 0 does nothing) and with all keys optional. Thus if there is any "data" key it will pre-empt on the data in the settings. I don't think this is really useful but it is OK. This piece of functionality can be re-introduced where it was removed also in a flat structure which better fits the optional nature of this "data" and is simpler to handle.

i.e. for AIS (in typescript notation):
```typescript
...
AISModActions: {
  tx?: number, // 0 or 1
  encode?: number, // 0 or 1
  data?: string
}
```

srcejon, 2021-11-18:
Great, thanks.

f4exb, 2021-11-27:
Deployed in v6.17.4

closed 2021-11-27 by f4exb