f4exb/sdrangel #1044: airports and heliports icons

choupette-change, 2021-11-12:
Hello, version 6 17 3 (Ubuntu 21.04 and Windows 10), the airports and heliports icons do not display on the ADSB Modulator plug-in map ??
[image]

choupette-change, 2021-11-12:
on the map feature too

srcejon, 2021-11-12:
You have the airport display distance at 1km - that means it will only display airports within 1km of your antenna position - change that to 100km or so.

srcejon, 2021-11-12:
Note that the Map feature doesn't currently display Airports - but I'm working on supporting it.

choupette-change, 2021-11-12:
I just tried 1, 10, 100, 1000 km nothing is displayed, it worked in previous versions (including 1 km), I have an airport 12 km from my home

srcejon, 2021-11-12:
Try redownloading the airport database

Check C:\Users\you\AppData\Roaming\f4exb\SDRangel\airportDatabase.csv - it should be about 10MB

choupette-change, 2021-11-12:
I have done this several times already and I cannot find this airportDatabase.csv file on Ubuntu (locate airportDatabase.csv) and Windows. But I think there is a pb, pressing the icon is very short and fast and I don't think a file is actually downloaded (problem de lien URL ?)

PR #1047 (2021-11-13): ADS-B updates
Support different map types (Aviation, Street and Satellite)
Add display of airspaces and NAVAIDs.
Display photo of highlighted aircraft.
Use latest database URLs to fix #1044

Note that this requires a new QT module: qml-module-qtgraphicaleffects. Patch includes updates to debian control and appveyor file.
closed 2021-11-13 by f4exb