f4exb/sdrangel #1049: Map display Adsb plugin (question or bug)

choupette-change, 2021-11-14:
With the Ubuntu version only, with ADS-B demodulator plugin, I often have display latencies (a few seconds(7,8 s)) of the plug-in channel map and feature map. But only with the vertical scrollbar (not mouse scroll wheeling, to zoom in or out). The map remains black or is partially displayed ??? I also saw it on previous versions. I have reinstalled SDRAngel several times on different PCs and I still have the same problem ???
[image]

srcejon, 2021-11-14:
This bit of code is entirely down to the Qt Location library, I think. Maps are downloaded in real-time from the Internet, so it could be a network related delay. Have a look in the log file and make sure there aren't any related warning messages.

choupette-change, 2021-11-14:
Thanks for your feedback but i have a high speed internet connection and it's ok with the windows version ?? Where is log file on windows and Ubuntu, please ??

srcejon, 2021-11-14:
Log file location is set under Preferences > Logging menu