f4exb/sdrangel #1050: Preset enhancements

srcejon, 2021-11-19:
Currently Device/Channel presets are separate from Feature presets. While this is nice in some use cases, in others, it would be nice if the two could be grouped as one. Perhaps there could be an option in the Device/Channel preset that determined whether it should save/restore Features as well?

Also, it would be useful if there were "application" presets, that included settings for multiple DeviceSets and also determined which Devices were used in each. E.g. for those with multiple SDRs, it would be nice if SDRangel started up with all DeviceSets setup to how there were on exit (As is currently the case for DeviceSet 0).

f4exb, 2021-11-20:
Regarding mixing deviceset (device+channels) and featureset presets in one display this is a big issue because they are very different in nature. Most significantly the deviceset presets are ordered by center frequency of the device. In order to combine both displays one would have to drop the frequency display and ordering in the deviceset presets

[image]

srcejon, 2021-11-20:
Yes, the request isn't for displaying two different types of preset in the same table, but having a single preset/configuration that sets device settings, channels and features all in one. 

So, for example,  we could have:
An AIS configuration that included two AIS demods, the AIS feature and the map.
A satellite voice configuration that included one RX deviceset with NFM demod, one TX deviceset with NFM mod, the simple PTT, sat tracker, rotator and map features.
A digital video config that includes both DATV Mod and Demod.
A NOAA config including an APT demod and Sat Tracker.
And so on.

This could be implemented by having a configuration that references an existing channel preset and feature preset, or similar, so you still have all existing functionality.

f4exb, 2021-11-20:
OK I see so in fact it all boils down to what I called the "second part". In fact the presets are not the point what you would like to do is to be able to save and retrieve a complete configuration with all its devicesets and presets. You do not necessarily have to use a specific interface to combine presets you just build the instance with or without presets and once you are satisfied with it you "save" it in order to be able to retrieve it later. You would also like to be able to save many configurations and have them listed somewhere more or less in the same format as the presets display.

For what concerns "loading" complete configurations with many devicesets and features this already exists using the REST API via a Python script: https://github.com/f4exb/sdrangel/tree/master/scriptsapi#configpy But you have to build the JSON configuration by hand. I already had in my mind to extend the `config.py` script functionnality so that it is also able to dump a SDRangel instance configuration in the same format so that it can be used to reload a configuration in a next session or in a server instance. I think that being able to use this script two ways would be useful anyway particularly for headless operations.

Note that it is possible to invoke the config,py script from the GUI using the "Commands". For example:

[image]

<details>
<summary>JSON file content</summary>

```json
[
    {
        "endpoint": "/deviceset",
        "method": "POST",
        "msg": "add Rx 1 device set"
    },
    {
        "endpoint": "/deviceset/1/device",
        "method": "PUT",
        "payload": {
            "hwType": "LocalInput"
        },
        "msg": "setup Local Input on Rx 1"
    },
    {
        "endpoint": "/preset",
        "method": "PATCH",
        "payload": {
            "deviceSetIndex": 1,
            "preset": {
                "groupName": "QO100",
                "centerFrequency": 489605000,
                "type": "R",
                "name": "Narrowband slave low"
            }
        },
        "msg": "load preset on Rx 1"
    },
    {
        "endpoint": "/deviceset",
        "method": "POST",
        "msg": "add Rx 2 device set"
    },
    {
        "endpoint": "/deviceset/2/device",
        "method": "PUT",
        "payload": {
            "hwType": "LocalInput"
        },
        "msg": "setup Local Input on Rx 2"                                                                                                                                                                                                  
    },                                                                                                                                                                                                                                      
    {                                                                                                                                                                                                                                       
        "endpoint": "/preset",                                                                                                                                                                                                              
        "method": "PATCH",                                                                                                                                                                                                                  
        "payload": {                                                                                                                                                                                                                        
            "deviceSetIndex": 2,                                                                                                                                                                                                            
            "preset": {                                                                                                                                                                                                                     
                "groupName": "QO100",                                                                                                                                                                                                       
                "centerFrequency": 489885000,                                                                                                                                                                                               
                "type": "R",                                                                                                                                                                                                                
                "name": "Narrowband slave high"                                                                                                                                                                                             
            }                                                                                                                                                                                                                               
[... cut ...]
So I think I will not reinvent the wheel here more re-implementing the `config.py` functionality inside the main program. Also the JSON file fornat is a better way to store such configurations because it can be edited by hand and also it can be reused for a headless instance or be transferred between computers. 

Lastly having a config fired up at startup time should only be an option. It would be very costly to change the default as it is today and also 90% if not more of the users run the program with a single receiver. Note that this was originally done just so that the GUI displays something and more conveniently the last receiver that was used. The server flavor starts empty.

srcejon, 2021-11-20:
Yep, sounds good. 

Being able to create a config in the GUI and then load it in the server would make it much easier to set up the server.

f4exb, 2022-02-17:
With the projected revamping of the UI the very nature of presets is going to change so I'd better postpone to the new UI implementation or make it part of the project.

f4exb, 2022-04-25:
done in v7.0.0-alpha.1

closed 2022-04-25 by f4exb