f4exb/sdrangel #1051: Reordering in preset save/load breaks links between plugins

srcejon, 2021-11-23:
We have several plugins that have settings that refer to other plugins. E.g. in Rotator Controller, you can specify the source plugin as R0:0 or R1:2 etc

Currently, when presets are saved or loaded (not sure which), the channels and features are reordered alphabetically. This means that the  settings that point to other plugins may no longer be correct.

f4exb, 2021-11-23:
This is at "save" time. For the Rx channels here: https://github.com/f4exb/sdrangel/blob/master/sdrbase/device/deviceset.cpp#L220 and similarly for Tx at line 321 and MIMO at line 424.

Similarly for features this is done here: https://github.com/f4exb/sdrangel/blob/master/sdrbase/feature/featureset.cpp#L182

So these calls have to be suppressed.

f4exb, 2021-11-24:
Well, seems it is also somewhere else but only for the UI ...
https://github.com/f4exb/sdrangel/blob/master/sdrgui/device/deviceuiset.cpp#L226
and
https://github.com/f4exb/sdrangel/blob/master/sdrgui/feature/featureuiset.cpp#L166

commit 9f39a1ab (2021-11-24): Removed ordering of channels and features in presets. Fixes #1051
commit 4b80fbf3 (2021-11-24): Remove sorting of channels and features when saving presets in GUI. Fixes #1051
closed 2021-11-24 by f4exb