f4exb/sdrangel #1062: is this a bug ?? plugin ADSB

choupette-change, 2021-11-27:
Hello, is this a bug ?? For example, if I position "My home" at the coordinates of Geneva airport (Geneva Cointrin International Airport) on the map "My home" is ok, but Geneva airport is badly positioned at 27.8 km from "My Home" ?? ? I took the coordinates in the "airportDatabase.csv" file. I installed SDRANGEL with the latest .deb. Several airports are positioned at the same place ?? Is it a km / mile conversion problem or Anglo-Saxon/European input (. or ,) ??? it's ok with the Windows version of SdrAngel. Thanks for your feedback.
[image]
[image]
[image]

srcejon, 2021-11-28:
There seems to be a small issue that when the map first appears, some icons are in the wrong location:

[image]

However, if you pan or zoom, then then immediately move to the correct place:

Here's LSGG centered on the runway:

[image]

I can't reproduce what is in your pictures.

srcejon, 2021-11-28:
I do notice your version of Qt is older than mine (5.12 vs 5.14) - and the mapping is part of Qt

srcejon, 2021-11-28:
When you say you installed from the latest .deb - what's the link to the file you used? Thanks.

f4exb, 2021-11-28:
The .deb in the "assets" section of the releases is for Ubuntu 20.04 and hence Qt 5.12.8

srcejon, 2021-11-28:
Thanks.

Works for me with Qt 5.12.8 and Ubuntu 20.04.03 LTS as well using the 6.17.4 .deb file.

choupette-change, 2021-11-28:
I had reinstalled a blank version of Ubuntu 20 04, .deb link : https://github.com/f4exb/sdrangel/releases/download/v6.17.4/sdrangel-1183-master.tar.gz. See my screenshots above for Ubuntu, Sdrangel and Qt versions.
I just reinstalled Sdrangel on another PC, by sources, SDRangel 6 17 4 and Qt 5 15 2 on Ubuntu 21 10 and I have exactly the same problem ???

srcejon, 2021-11-29:
> Several airports are positioned at the same place ?? Is it a km / mile conversion problem or Anglo-Saxon/European input (. or ,) ?

Yes, looks like a , . problem due to the value of LC_NUMERIC.

closed 2021-11-29 by f4exb

choupette-change, 2021-11-29:
I tested the latest version it's ok for me :)
[image]

PR #1065 (2021-11-29): Fix #1062 - Use C locale when parsing airport/navaid files
Fixes #1062 and #732

Airport locations in ADS-B plugin were at wrong locations for users with locales where decimal point is , rather than . This should fix that, by using C locale when parsing the .csv files.

Also, updates VOR demod GUI so more than one row is visible.
commit 259e0ad6 (2021-11-29): Fix #1062 - Always use C locale for parsing coords in airport database
commit a8d81b14 (2021-11-29): Merge pull request #1065 from srcejon/fix_1062

Fix #1062 - Use C locale when parsing airport/navaid files